dora-rs/dora #93: Plans for command line interface

phil-opp, 2022-10-05:
We are planning a dora command line tool with the following subcommands:

- [x] `check`: Pre-check the dora installation/environment for potential problems (e.g., YAML syntax)
- [x] `build`: Build the dora nodes/operators (similar to run.rs in current examples)
- [ ] `graph`: Generate dataflow graph per YAML spec
  - [x] visualize as mermaid code
  - [ ] visualize as image directly (e.g. as `svg`)
  - [x] open image automatically, e.g. in OS or browser window
- [x] `help`: Help about any command (generated by clap)
- [ ] `templates`: Dora node/operator template
  - [x] Rust
  - [x] C
  - [ ] Python
  - [ ] C++
- [ ] `start`: Run dora dataflow
- [ ] `stop`: Stop dora dataflow
- [ ] `up`: Install all dora dependencies and start dora coordinator, iceoryx daemon and/or zenoh router etc
- [ ] `destroy`: Cleanup and destroy all dora related processes
- [ ] `dashboard`: Open the dora dashboard in a web browser (future)
- [ ] `logs`: Tail logs from customer node/operators
- [ ] `metrics`: Fetch metrics from customer node/operators
- [ ] `stats`: Display dora/traffic statistics 
- [ ] `list`: List running dora entities (resources)
- [ ] `get`: Get info about the dora entity (future may need CRUD on dora entity)
- [ ] `upgrade`: Upgrade dora 

See the discussion in #81 for more context.

elBoberido, 2022-11-06:
@phil-opp creating an API to check whether the iceoryx daemon is running is on our todo list. We are at the moment just busy with other stuff.

phil-opp, 2022-11-07:
Great to hear that, thanks! For now we just check the running processes, so we have a workaround.

closed 2026-01-30 by haixuanTao