dora-rs/dora #136: Stop custom node when stopping the dataflow

phil-opp, 2022-11-09:


phil-opp, 2022-11-09:
@haixuanTao Could you give more details on this request?

haixuanTao, 2022-11-09:
The problem is that some python program such as ROS python nodes are based on a forever loop. When starting them through dora we might want to stop them on `dora stop`.

haixuanTao, 2022-11-09:
But after discussion, we chose to leave the closing of those forever loops to the user as we might not have enough information on how to stop the nodes.

phil-opp, 2023-03-09:
With the new event stream design merged in #162, all nodes should listen for incoming events, even if they don't expect inputs. Nodes that don't do this will be killed as soon as #184 is merged.

haixuanTao, 2024-08-31:
Closing as rhis is now implemented with grace duration

closed 2024-08-31 by haixuanTao