dora-rs/dora #150: Rust example build error when using 0.1.0 instruction

heyong4725, 2022-11-22:
Environment: Ubuntu 22.04

Follow the README, using Rust template to build rust example (default), I have the following error (why we have iceoryx dependecy?):

(base) demo@:~/src/dora-rs/releases/0.1.0/rust_proj$ cargo build -p node_1
warning: skipping duplicate package `rust-dataflow-example-sink` found at `/home/demo/.cargo/git/checkouts/dora-865db9b4eb79312d/c9f4a1a/examples/rust-dataflow/sink`
   Compiling iceoryx-sys v0.1.0 (https://github.com/eclipse-iceoryx/iceoryx-rs.git#68fbd034)
error: failed to run custom build command for `iceoryx-sys v0.1.0 (https://github.com/eclipse-iceoryx/iceoryx-rs.git#68fbd034)`

Caused by:
  process didn't exit successfully: `/home/demo/src/dora-rs/releases/0.1.0/rust_proj/target/debug/build/iceoryx-sys-6b251d9e411e136f/build-script-build` (exit status: 1)
  --- stdout
  Output { status: ExitStatus(unix_wait_status(0)), stdout: "", stderr: "HEAD is now at f756b7c99 Merge pull request #1327 from ApexAI/iox-#1323-set-version-to-2.0.2\n" }
  --- stderr
  CMake Error: The current CMakeCache.txt directory /home/demo/src/dora-rs/releases/0.1.0/rust_proj/target/debug/build/iceoryx-sys-ac30c768c6b445f0/out/iceoryx-build/iceoryx_hoofs/CMakeCache.txt is different than the directory /home/demo/src/dora-rs/releases/0.1.0/c_proj/target/debug/build/iceoryx-sys-ac30c768c6b445f0/out/iceoryx-build/iceoryx_hoofs where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
  CMake Error: The source "/home/demo/src/dora-rs/releases/0.1.0/rust_proj/target/debug/build/iceoryx-sys-ac30c768c6b445f0/out/iceoryx-git/iceoryx_hoofs/CMakeLists.txt" does not match the source "/home/demo/src/dora-rs/releases/0.1.0/c_proj/target/debug/build/iceoryx-sys-ac30c768c6b445f0/out/iceoryx-git/iceoryx_hoofs/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
  Error: Custom { kind: Other, error: "Could not run cmake for 'iceoryx_hoofs'!" }

I have cleaned .cargo/git cache by "cargo cache -a" command, don't know why it still pull in iceoryx and examples/rust-dataflow/sink dependencies

heyong4725, 2022-11-22:
@phil-opp can you take a look?

haixuanTao, 2022-11-29:
On a fresh docker of Ubuntu 22 with bare minimum dependency. I was not able to reproduce your error. 
The dependency I had to add from a bare minimal Ubuntu were:
```bash
apt-get install git 
apt-get install libacl1-dev
apt install libstdc++-10-dev libstdc++-10-doc
apt install cmake
apt install build-essential
``` 

I had a silent bug with `iceoryx` when I lacked `git` and no error showed up. So something like this might be the cause f the issue.

heyong4725, 2022-12-07:
The issue is fixed.

closed 2022-12-07 by heyong4725