dora-rs/dora #206: exec the command `dora upgrade`  exception

meua, 2023-03-10:
**Describe the bug**
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Dora upgrade daemon: `dora upgrade` 
2. The error is as follows：
thread 'main' panicked at 'not yet implemented', binaries/cli/src/main.rs:148:29
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

**Screenshots or Video**
[image]

**Environments (please complete the following information):**
 - System info: 
 Linux jia 5.15.0-67-generic #74~20.04.1-Ubuntu SMP Wed Feb 22 14:52:34 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
 - Dora version：
 dora-cli 0.1.3

meua, 2023-03-10:
dora dashboard
dora upgrade
dora logs 
dora metrics
dora stats

meua, 2023-03-10:
The same problem exists with the following command：
`dora dashboard`
`dora upgrade`
`dora logs `
`dora metrics`
`dora stats`

phil-opp, 2023-03-10:
Thanks a lot for reporting! These are all commands that we envisioned for future releases of dora (see #81). However, they are not designed or implemented right now. So I think that it's best to remove them for now. I'll prepare a PR.

PR #207 (2023-03-10): Remove unimplemented CLI arguments for now
We planned these additional commands in #81, but they are not implemented right now. I think it's best to remove them for now to avoid confusion.

Fixes #206
closed 2023-03-10 by haixuanTao