dora-rs/dora #233: Add `Event::OutputReady` to the Event stream

haixuanTao, 2023-03-23:
**Is your feature request related to a problem? Please describe.**
Make sure that when sending output the nodes or operators are ready to receive those inputs.

**Current solution**
Current solution is quite hacky and requires to make a loop, like:
```yaml
nodes:
  - id: oasis_agent
    custom:
      inputs:
        carla_gps_ready: carla_gps_op/ready
      outputs:
        - check
      source: shell

  - id: carla_gps_op
    operator:
      python: ../../carla/carla_gps_op.py
      outputs:
        - ready
      inputs:
        check: oasis_agent/check
```

**Describe the solution you'd like**

### For Core / Daemon
- [ ] Add `Event::OutputReady` to our event stream
- [ ] Send `NodeEvent::OutputReady { id }` to signal to a node that an output is ready
- [ ] Send `NodeEvent::AllOutputsReady` to signal that all outputs are ready ( similar to `NodeEvent::AllInputsClosed` )

### For Runtime / Operators
- [ ] Send a `Ready` event to the daemon when it has finished its initialization method (e.g.: `__init__` for python).
> Notes: This is transparent to the user 
 
### For Custom Nodes
- [ ] Add a method `.ready()` or `.send_event(Ready)` to the node API to send event to the daemon 
> Notes: this requires that the user add `.ready()` in its code. 
 

Open for input :smile:

phil-opp, 2023-03-23:
I prepared a PR at https://github.com/dora-rs/dora/pull/236. Instead of introducing a new `Ready` message, I reused the existing `Subscribe` message that all nodes send on initialization. By delaying the reply to this message until all nodes are ready, we can start them at the same time.

haixuanTao, 2023-03-24:
That's awesome! I will write more Issues before implementation. Thanks a lot!

PR #236 (2023-03-24): Synchronize node and operator start
- Synchronize startup of nodes through subscribe request
  - Nodes wait for a reply after sending a subscribe request.
  - By delaying that reply until all nodes are subscribed, we can synchronize the start time of the dataflow.
- Only start timer tasks once dataflow is ready to start
  - This way, we don't fill the queue with tick events that occured before the synchronized start happened.
- dora-runtime: Only subscribe to daemon once all operators are ready
  - The `dora-runtime` uses the normal node API, so it uses the same synchronization system.
  - By  [... cut]
closed 2023-03-24 by haixuanTao