dora-rs/dora #250: The dora service is not started. An error is reported execute dora list

meua, 2023-04-14:
**Describe the bug**
The dora service is not started. An error is reported execute dora list

**To Reproduce**
Steps to reproduce the behavior:
1. dora list

**Screenshots or Video**
[image]

**Environments (please complete the following information):**
 - System info: `Linux jia 5.15.0-69-generic #76~20.04.1-Ubuntu SMP Mon Mar 20 15:54:19 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`
 - Dora version : 0.2.2

phil-opp, 2023-04-17:
Could you clarify what behavior you are expecting? Do you think that the coordinator and daemon should be started automatically on `dora list`? Or do mean that we should improve the error message to e.g. "no dora coordinator is running" instead of "Connection refused"?

meua, 2023-04-18:
> Could you clarify what behavior you are expecting? Do you think that the coordinator and daemon should be started automatically on `dora list`? Or do mean that we should improve the error message to e.g. "no dora coordinator is running" instead of "Connection refused"?

No, I think that when dora is not started, execute the dora list command, it should not report an error, it should remind the prompt that dora is not started

phil-opp, 2023-04-18:
Thanks for clarifying! I opened https://github.com/dora-rs/dora/pull/254 to fix this.

haixuanTao, 2023-04-18:
I may be wrong, but is this issue more geared toward the way the error is reported rather than the actual content of the error @meua ?

meua, 2023-04-19:
> I may be wrong, but is this issue more geared toward the way the error is reported rather than the actual content of the error @meua ?

The dora return should clearly explain the cause of the problem, and should not directly throw an exception

PR #254 (2023-04-19): CLI: Improve error messages when coordinator is not running
This PR improves the reported errors when the CLI fails to connect to dora daemon. Instead of a "connection refused" error, we now output a "could not connect to dora coordinator" error.

Fixes #250
closed 2023-04-19 by phil-opp