dora-rs/dora #252: The `dora stop uuid` unable to delete a running dataflow

meua, 2023-04-18:
**Describe the bug**
The `dora stop uuid` unable to delete a running dataflow. The `dora destroy `command works, but an exception is reported, and the exception is not handled correctly

**To Reproduce**
Steps to reproduce the behavior:
1. Dora start daemon: `dora up` 
2. then execute `dora start graphs/tutorials/webcam_yolov5.yaml --attach `

**Screenshots or Video**
[image]

**Environments (please complete the following information):**
 - System info: Linux jia 5.15.0-69-generic #76~20.04.1-Ubuntu SMP Mon Mar 20 15:54:19 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
 - Dora version: 0.2.2

phil-opp, 2023-04-18:
Thanks for reporting! Could you give us more details on how to reproduce this issue? The screenshot shows a different set of commands than you listed above. And I assume the dataflow you're referencing is https://github.com/dora-rs/dora-drives/blob/main/graphs/tutorials/webcam_yolov5.yaml?

meua, 2023-04-19:
> Thanks for reporting! Could you give us more details on how to reproduce this issue? The screenshot shows a different set of commands than you listed above. And I assume the dataflow you're referencing is https://github.com/dora-rs/dora-drives/blob/main/graphs/tutorials/webcam_yolov5.yaml?

Yes! just do it

meua, 2023-05-15:
This problem, has been fixed in this commit `a308c48f3e842cecd994a9b36a51750e008e63c1`.
```bash
git show a308c48f3e842cecd994a9b36a51750e008e63c1
```

closed 2023-05-15 by meua