dora-rs/dora #253: Accidentally dora command is unresponsive and stuck

meua, 2023-04-18:
**Describe the bug**
Accidentally dora command is unresponsive and stuck

**To Reproduce**
Steps to reproduce the behavior:
1. Few cases，when executing command `dora up` , `dora start dataflow.yaml`,  `dora stop`, `dora destroy`

**Environments (please complete the following information):**
 - System info: Linux jia 5.15.0-69-generic #76~20.04.1-Ubuntu SMP Mon Mar 20 15:54:19 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
 - Dora version：0.2.2

phil-opp, 2023-04-18:
Could you give us more details on how to reproduce this issue?

haixuanTao, 2023-04-19:
I think I can reproduce the issue with:
```
dora up
dora destroy
dora up # <--- This hangs
```

I think it is due the coordinator waiting for something which makes it unable to respond to other request.

phil-opp, 2023-04-19:
Hmm, I tried it multiple times but I couldn't reproduce the issue on the `main` branch. Which dora version are you using @haixuanTao ?

haixuanTao, 2023-04-19:
Yep, I think, I will investigate on my end

meua, 2023-04-20:
[image]
reproduce the problem again, the steps are as shown above.

The premise is that dora start has an exception, as shown below:
[image]

meua, 2023-04-20:
@phil-opp

meua, 2023-04-21:
the service 'dora daemon' hung up.

haixuanTao, 2023-04-21:
I think it's probably linked to the operator yolov5 not accessing github, stucking the initialisation function. But it's going to be very hard for Philipp to reproduce.

haixuanTao, 2023-04-25:
Having retested this issue, this is the stack trace:

```bash
(base) ~/D/C/dora ❯❯❯ RUST_LOG=trace dora destroy                                           (base) fix-coordinator-loop ✭
  2023-04-25T08:21:11.484181Z TRACE dora_coordinator::control: Control connection closed
    at binaries/coordinator/src/control.rs:90

  2023-04-25T08:21:11.484197Z TRACE dora_coordinator: Handling event Control(IncomingRequest { request: Destroy, reply_sender: Sender { inner: Some(Inner { state: State { is_complete: false, is_closed: false, is_rx_task_set: true, is_tx_task_set: false } }) } })
    at binaries/coordinator/src/lib.rs:142

  2023-04-25T08:21:11.484227Z  INFO dora_coordinator: Received destroy command
    at binaries/coordinator/src/lib.rs:403

  2023-04-25T08:21:11.484359Z  INFO dora_daemon: received destroy command -> exiting
    at binaries/daemon/src/lib.rs:331
    in dora_daemon::run_inner with self.machine_id: 

Send destroy command to dora-coordinator
  2023-04-25T08:21:11.484604Z TRACE dora_coordinator::control: Control connection closed
    at binaries/coordinator/src/control.rs:90
```

It seems to be due to this: `Control connection closed` which happens because there is an `ErrorKind::UnexpectedEof` 

Looking at running process the `dora daemon` has exited.

This is probably linked to an error on sending a confirmation of the dora daemon to the coordinator to have been successfully destroyed.

phil-opp, 2023-04-26:
> It seems to be due to this TRACE: `Control connection closed` which happens because there is an `ErrorKind::UnexpectedEof`

This is expected, as the CLI closes it's control connection to the coordinator when it exits.

phil-opp, 2023-04-26:
> The premise is that dora start has an exception, as shown below: [image]

This seems to be the real issue here. The python operator seems to require GLIBCXX_3.4.29 (required by matplotlib), but it is not found. This error brings down the whole runtime node. I'm not sure why the daemon does not detect this error, but my guess is that it is stuck waiting for the node to finish initialization (for the synchronized start introduced in #236).

So I think there are two things that we need to look into:

- Why is the mentioned GLIBCXX version required, but not found?
- Why doesn't the dora daemon detect the operator/node initialization error?

haixuanTao, 2023-04-27:
I opened #271 to track this issue: Why doesn't the dora daemon detect the operator/node initialization error?

phil-opp, 2023-06-21:
Does this issue still happen on the latest version (i.e. with #271 merged)?

meua, 2023-06-27:
> Does this issue still happen on the latest version (i.e. with #271 merged)?

The situation described above has not happened again, but there is still a situation where dora stop cannot stop dataflow. This problem occurs because an exception occurs inside an operator that dataflow depends on, as shown below:
```bash
(dora3.7) jarvis@jia:~/coding/pyhome/github.com/dora-rs/dora-drives$ dora list
Running dataflows:
- [YOLOv8] 4aba7bb7-7966-4839-921d-72c575f7ea33
(dora3.7) jarvis@jia:~/coding/pyhome/github.com/dora-rs/dora-drives$ dora stop 
> Choose dataflow to stop: [YOLOv8] 4aba7bb7-7966-4839-921d-72c575f7ea33
(dora3.7) jarvis@jia:~/coding/pyhome/github.com/dora-rs/dora-drives$ dora list
Running dataflows:
- [YOLOv8] 4aba7bb7-7966-4839-921d-72c575f7ea33
(dora3.7) jarvis@jia:~/coding/pyhome/github.com/dora-rs/dora-drives$ dora -V
dora-cli 0.2.3
(dora3.7) jarvis@jia:~/coding/pyhome/github.com/dora-rs/dora-drives$ 

```

vi webcam_yolov8.yaml
```yaml
nodes:
  - id: webcam
    operator:
      python: ../../operators/webcam_op.py
      inputs:
        tick: dora/timer/millis/100
      outputs:
        - image
    env:
      DEVICE_INDEX: 2

  - id: yolov8
    operator: 
      outputs:
        - bbox
      inputs:
        image: webcam/image
      python: ../../operators/yolov8_op.py
    env:
      PYTORCH_DEVICE: "cuda"
#      YOLOV8_PATH: $DORA_DEP_HOME/dependencies/YOLOv8/
#      YOLOV8_WEIGHT_PATH: $DORA_DEP_HOME/dependencies/YOLOv8/weights/yolov8n.pt

  - id: plot
    operator:
      python: ../../operators/plot.py
      inputs:
        image: webcam/image
        obstacles_bbox: yolov8/bbox
```
```bash
(dora3.7) jarvis@jia:~/coding/pyhome/github.com/dora-rs/dora-drives$ RUST_LOG=true dora start graphs/tutorials/webcam_yolov8.yaml --attach --hot-reload --name YOLOv8
4aba7bb7-7966-4839-921d-72c575f7ea33
(dora3.7) jarvis@jia:~/coding/pyhome/github.com/dora-rs/dora-drives$ 
```

meua, 2023-06-27:
> Does this issue still happen on the latest version (i.e. with #271 merged)?

v0.2.3 problem still exists, dora-cli unresponsive.
```bash
(dora3.7) jarvis@jia:~/coding/pyhome/github.com/dora-rs/dora-drives$ dora destroy
  2023-06-27T09:32:54.865121Z  WARN dora_daemon::node_communication: failed to send event to daemon

Location:
    /home/runner/work/dora/dora/binaries/daemon/src/node_communication/mod.rs:490:26
    at binaries/daemon/src/node_communication/mod.rs:253

  2023-06-27T09:32:54.865152Z  WARN dora_daemon::node_communication: failed to receive reply from daemon

Location:
    /home/runner/work/dora/dora/binaries/daemon/src/node_communication/mod.rs:494:30
    at binaries/daemon/src/node_communication/mod.rs:253

Send destroy command to dora-coordinator
(dora3.7) jarvis@jia:~/coding/pyhome/github.com/dora-rs/dora-drives$ dora list
   
```

**To Reproduce**
Steps to reproduce the behavior:
1. Dora start daemon: `dora up` 
2. Start a new dataflow: `dora start graphs/tutorials/webcam_yolov5.yaml --attachml --attach`

**Screenshots or Video**
[image]
[image]

**Environments (please complete the following information):**
 - System info: `ubuntu 20.04 LTS`
 - Dora version: `v0.2.3`

closed 2024-10-08 by haixuanTao