dora-rs/dora #260: The process cannot be destroyed by the command `dora destroy`

meua, 2023-04-21:
**Describe the bug**
The process cannot be destroyed by the command `dora destroy`

**To Reproduce**
Steps to reproduce the behavior:
1. Dora start daemon: `dora up` 
2. Destroy dataflow: `dora destroy`

**Screenshots or Video**
[image]

**Environments (please complete the following information):**
 - System info: ubuntu 20.04
 - Dora version: 0.2.2

haixuanTao, 2023-04-25:
Can you redo the exact same command but setting `export RUST_LOG=trace` before running them?

meua, 2023-04-28:
> Can you redo the exact same command but setting `export RUST_LOG=trace` before running them?

Ok, I got it, I'll do it and give you feedback.

haixuanTao, 2024-08-31:
I think this is a duplicate of #253

closed 2024-08-31 by haixuanTao