dora-rs/dora #261: Dora adds support for golang

meua, 2023-04-23:
**Is your feature request related to a problem? Please describe.**
At present, dora supports c++/c/python/rust language, and has basically supported several commonly used languages, but now the mainstream of cloud native needs to be golang, so some AI algorithms will use golang to encapsulate

**Describe the solution you'd like**
dora supports golang to define operator/node

**Describe alternatives you've considered**
Use golang to implement dora operator or node, and run it successfully

haixuanTao, 2023-04-25:
We currently have no plan in supporting golang. I have opened a PR to clarify our planned feature.

phil-opp, 2023-04-26:
I'm not familar with golang, but I think it should be possible to use `cgo` to bridge to one of our existing libraries (e.g. C or Rust). See https://github.com/mediremi/rust-plus-golang for some general examples on how to call Rust code from golang.

I don't think that there is a technical reason that prevents us from adding golang support to dora, it's just not a priority right now. We would be happy to merge a PR for this though.

closed 2023-04-26 by phil-opp