dora-rs/dora #262: todo: dora provides Chinese documents

meua, 2023-04-23:
**Is your feature request related to a problem? Please describe.**
Dora currently only has English documents and no Chinese documents, which is not very friendly to Chinese domestic users

**Describe the solution you'd like**
Dora documents can be switched between Chinese and English

**Describe alternatives you've considered**
dora supports Chinese documents

meua, 2023-05-08:
I have translated the English technical documentation of dora into the corresponding Chinese technical documentation. How should I deploy it to facilitate language switching?

haixuanTao, 2023-05-08:
I'm working on making dora doc multi-lingual although this feature is not ready yet.

meua, 2023-05-08:
> I'm working on making dora doc multi-lingual although this feature is not ready yet.

:+1: :+1: :+1:

haixuanTao, 2023-05-26:
The new documentation website: dora.carsmos.ai now supports Chinese translation through crowdin.

closed 2023-05-26 by haixuanTao