dora-rs/dora #266: Dataflow with no specified name shows unnamed

meua, 2023-04-24:
**Is your feature request related to a problem? Please describe.**
Dataflow with no specified name shows unnamed, as follows:
[image]
```
impl Display for DataflowId {
    fn fmt(&self, f: &mut std::fmt::Formatter<'_>) -> std::fmt::Result {
        if let Some(name) = &self.name {
            write!(f, "[{name}] {}", self.uuid)
        } else {
            write!(f, "[<unnamed>] {}", self.uuid)
        }
    }
}
```

**Describe the solution you'd like**
Is it possible for dataflow without a name to randomly generate a unique name for it.

phil-opp, 2023-04-26:
Thanks for reporting! Could you clarify what kind of name we should generate? There is already the unique ID for each dataflow (shown after the `[<unnamed>]`).

phil-opp, 2023-04-26:
I guess we could use a crate such as [`names`](https://docs.rs/names/latest/names/) to generate human-readable names that are easier to read than the UUIDs. Is this what you were thinking?

meua, 2023-04-28:
> 
Yes, short names are easier to remember, and using a combination of letters and numbers can significantly shorten this unique identifier

PR #276 (2023-05-11): Autogenerate a name for started dataflows if none is given
Generates random `adjective-noun` names such as `faded-eye` or `grubby-bath`.

Fixes #266
commit 2401e434 (2023-05-11): Merge pull request #276 from dora-rs/assign-short-names

Autogenerate a name for started dataflows if none is given

Fixes #266
closed 2023-05-11 by meua