dora-rs/dora #267: Error: failed to send start dataflow message

NavyHub, 2023-04-26:
**Describe the bug**
```bash
(dora3.7) jarvis@jia:~/coding/dora_home/dora-drives$ proxychains dora start graphs/tutorials/webcam_single_dpt_frame.yaml --attach --hot-reload --name dpt_midas
ProxyChains-3.1 (http://proxychains.sf.net)
|S-chain|-<>-127.0.0.1:1080-<><>-127.0.0.1:6012-<><>-OK
Error: failed to send start dataflow message

Caused by:
    failed to fill whole buffer

Location:
    binaries/cli/src/main.rs:214:10
(dora3.7) jarvis@jia:~/coding/dora_home/dora-drives$ 

```

**To Reproduce**
Steps to reproduce the behavior:
1. Dora start daemon: `dora up` 
2. Start a new dataflow: `proxychains dora start graphs/tutorials/webcam_single_dpt_frame.yaml --attach --hot-reload --name dpt_midas`

**Expected behavior**
dora can support proxy mode, the purpose is to quickly download foreign acceleration algorithm models

**Environments (please complete the following information):**
 - System info: `Linux jia 5.15.0-70-generic #77~20.04.1-Ubuntu SMP Wed Apr 5 09:38:34 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`
 - Dora version: 0.2.2

haixuanTao, 2023-04-26:
We don't plan on supporting adhoc proxy for now. You can check your proxy configuration for whitelisting localhost and `127.0.0.1`.

closed 2023-04-26 by haixuanTao

phil-opp, 2023-04-26:
I would be curious to understand why this is failing. @NavyHub Could you explain a bit more what proxychains does? I'm not familiar with this tool.