dora-rs/dora #283: error[E0554]: `#![feature]` may not be used on the stable release channel

meua, 2023-05-06:
**Describe the bug**
```bash
error[E0554]: `#![feature]` may not be used on the stable release channel
  --> /home/jarvis/.cargo/registry/src/github.com-1ecc6299db9ec823/rustix-0.36.9/src/lib.rs:99:26
   |
99 | #![cfg_attr(rustc_attrs, feature(rustc_attrs))]
   |                          ^^^^^^^^^^^^^^^^^^^^
```

**To Reproduce**
Steps to reproduce the behavior:
1. Debug start coordinator.

**Environments (please complete the following information):**
 - System info: ubuntu 22.04
 - Dora version: dora-rs/dora main branch

sunfishcode, 2023-05-06:
This is fixed in rustix 0.36.11 and later, as well as 0.37.5 and later.

meua, 2023-05-07:
> This is fixed in rustix 0.36.11 and later, as well as 0.37.5 and later.

Thank you very much for the tip, I will try to do what you said.

commit dabc6742 (2023-05-08): Merge pull request #285 from meua/main

To solve this bug https://github.com/dora-rs/dora/issues/283, unify t…

Fix #283
closed 2023-05-08 by haixuanTao