dora-rs/dora #313: Cannot exec dora or dora-cli command when install dora with conda on macOS

jinlong-hao, 2023-07-04:
**Describe the bug**
cannot exec dora or dora-cli command when install dora with conda on macOS

**To Reproduce**
Steps to reproduce the behavior:

1. create a new conda env :  conda create -n dora python=3.9
2. activate dora:  conda activate dora
3. install dora and dora-cli in conda env:  pip install dora dora-cli 
4. exec dora-cli or dora comman in terminal

**Expected behavior**

expect dora can successfully run.

jinlong-hao, 2023-07-06:
It's my first time use rust. 
I installed rust and dora with cargo,  the problem solved.

closed 2023-07-06 by jinlong-hao

haixuanTao, 2023-07-06:
I might add the cli to pip so that people can install it directly without cargo. Have fun trying out Rust ☺️

haixuanTao, 2023-07-06:
For future reference, [`dora-cli`](https://pypi.org/project/dora-cli/) on Pypi is not our project.