dora-rs/dora #322: some errors when i run my code

Latitude9527, 2023-07-19:
I had run the webcam successfully,but when i run my driver about D435i,it showed errors  as follows:
          2023-07-19T06:52:02.167239Z ERROR dora_daemon: 
        bd218eb3-f987-4b59-92d6-b7fda991a2c6/webcam failed with exit code 1.

       Check logs using: dora logs bd218eb3-f987-4b59-92d6-b7fda991a2c6 webcam
                            
        at binaries/daemon/src/lib.rs:906

     2023-07-19T06:52:02.167258Z  WARN dora_daemon::pending: node `webcam` exited before initializing dora connection
        at binaries/daemon/src/pending.rs:76

       2023-07-19T06:52:02.376402Z ERROR dora_daemon: 
         bd218eb3-f987-4b59-92d6-b7fda991a2c6/plot failed with exit code 1.

        Check logs using: dora logs bd218eb3-f987-4b59-92d6-b7fda991a2c6 plot
                            
         at binaries/daemon/src/lib.rs:906
 
when i run the first comend 'dora logs bd218eb3-f987-4b59-92d6-b7fda991a2c6 webcam' ,it   showed a  link like                 "/home/latitude/Downloads/dora-drives-main/testD435i/webcam_op.py", but when i clicked it showed that  ' dora_input["metadata"] ' is
unexpected positional argument in 
                   send_output(
            "color_image",
            pa.array(color_image.ravel().view(np.uint8)),
            dora_input["metadata"],
        )
      i had tried to delete 'dora_input["metadata"]',but the same hints appers in 'pa.array(color_image.ravel().view(np.uint8)),'and  when i run the seond comand 'dora logs bd218eb3-f987-4b59-92d6-b7fda991a2c6 plot',the output is null.

haixuanTao, 2023-07-19:
Do you mind sharing the code that you're using for the webcam?

Latitude9527, 2023-07-20:
> 

> Do you mind sharing the code that you're using for the webcam?

of course ，my code are as follows:
from typing import Callable
from dora import DoraStatus
import os
import cv2
import numpy as np
import pyarrow as pa
import pyrealsense2 as rs

pa.array([])

OUTPUT_WIDTH = 640
OUTPUT_HEIGHT = 480
# DEVICE_INDEX = os.environ.get('DEVICE_INDEX', '0')

class Operator:
    """
    Compute a `control` based on the position and the waypoints of the car.
    """

    def __init__(self):
        self.pipeline = rs.pipeline()
        self.config = rs.config()
        self.config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 15)
        self.config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 15)
        self.pipeline.start(self.config)
        self.align_to = rs.stream.color
        self.align = rs.align(self.align_to)

    def on_event(
        self,
        dora_event: dict,
        send_output: Callable[[str, bytes], None],
    ) -> DoraStatus:
        if dora_event["type"] == "INPUT":
            return self.on_input(dora_event, send_output)
        return DoraStatus.CONTINUE

    def on_input(
        self,
        dora_input: dict,
        send_output: Callable[[str, bytes], None],
    ):

            frames = self.pipeline.wait_for_frames()

           ret, frames = self.pipeline.wait_for_frames()
           if ret:
                 color_frame = frames.get_color_frame()
                 depth_frame = frames.get_depth_frame()

                frames = cv2.resize(frames, (640, 480))
                frames = np.array(cv2.cvtColor(frames, cv2.COLOR_BGR2BGRA))
                frames = frames.astype(np.uint8)
                color_frame = frames.get_color_frame()
                #depth_frame = frames.get_depth_frame()
               # infrared_frame=frames.get_infrared_frame(1)

               color_image = np.asanyarray(color_frame.get_data())
        # depth_image = np.asanyarray(depth_frame.get_data())
        # infrared_image = np.asanyarray(infrared_frame.get_data())

        # depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        # # Stack both images horizontally
        # images = np.hstack((color_image, depth_colormap, infrared_image))
        # Add your image processing here (e.g., resizing, color conversion, etc.)
        # For example:
        # processed_frame = cv2.resize(color_image, (OUTPUT_WIDTH, OUTPUT_HEIGHT))
        # processed_frame = cv2.cvtColor(processed_frame, cv2.COLOR_BGR2BGRA)
        # Send the processed frame as output
        send_output(
            "color_image",
            pa.array(color_image.ravel().view(np.uint8)),
            dora_input["metadata"],
        )
        # send_output(
        #     "depth_image",
        #     pa.array(depth_image.ravel().view(np.uint8)),
        #     dora_input["metadata"],
        # )
        # send_output(
        #     "infrared_image",
        #     pa.array(infrared_image.ravel().view(np.uint8)),
        #     dora_input["metadata"],
        # )

        # Send the depth image as output
        # send_output(
        #     "depth_image",
        #     pa.array(depth_image.ravel().view(np.uint16)),
        #     dora_input["metadata"],
        # )

        return DoraStatus.CONTINUE

    def drop_operator(self):
        self.pipeline.stop()

Latitude9527, 2023-07-20:
> Do you mind sharing the code that you're using for the webcam?

I seem to have understood something. I got some inspiration from ROS2, where video streams need to be converted into formats required by publishers. Does Dora have such a tool?

haixuanTao, 2023-07-20:
Has your problem been solved? 

> I seem to have understood something. I got some inspiration from ROS2, where video streams need to be converted into formats required by publishers. Does Dora have such a tool?

The output seems to be in the right format currently. What do you mean by a tool for converting format?

closed 2023-07-20 by haixuanTao