dora-rs/dora #327: Ubuntu18.04 mismatch the lastest Dora

ChichengZZZ, 2023-07-27:
**Describe the bug**  

We want to install Dora(v0.2.3) on Nvidia Jetson nano, but Our development board can only be adapted to ubuntu 18.04, so we installed dora in the base environment of Ubuntu 18.04 (python3.8.8), but an error was reported ’illegal instruction‘，we think this issue is due to inappropriate python version then we install anaconda to create a virtual environment for python3.7.0. However, the problem has not been solved, and the original error is still reported as 'illegal instruction'. Please help me...

meua, 2023-07-27:
Hi, @haixuanTao xavier, I remember you installed dora on aarch64 before, can you help me to look at this problem： how to install python3.7 on aarch64.
```bash
agilex@agilex:~$ pyenv install 3.7.17
Downloading Python-3.7.17.tar.xz...
-> https://www.python.org/ftp/python/3.7.17/Python-3.7.17.tar.xz
error: failed to download Python-3.7.17.tar.xz

BUILD FAILED (Ubuntu 18.04 using python-build 20180424)

Results logged to /tmp/python-build.20230727161802.12549.log

Last 10 log lines:
/tmp/python-build.20230727161802.12549 ~
/home/agilex/.pyenv/plugins/python-build/bin/python-build: line 404: 12587 **Illegal instruction     (core dumped)** curl -q -o "${2:--}" -sSLf ${CURL_OPTS} "$1"
agilex@agilex:~$ python -V
Python 3.8.8
agilex@agilex:~$ pip install dora-rs=v0.2.4
Defaulting to user installation because normal site-packages is not writeable
^CERROR: Operation cancelled by user
agilex@agilex:~$
```

haixuanTao, 2023-07-27:
You should be able to install dora on all python version above 3.7. 

It seems that the illegal instruction comes from failing to build python for some reason. 

Can you try to download the following release: https://github.com/dora-rs/dora/releases/download/v0.2.4/dora_rs-0.2.4-cp37-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl

And install it with: pip install dora_rs-0.2.4-cp37-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl` 

You can also try to use a mirror when using pip if your in china such as https://gist.github.com/schnell18/d0ed716917905d2c142a370906cfa32f . That might be the issue.

meua, 2023-07-28:
> You should be able to install dora on all python version above 3.7.
> 
> It seems that the illegal instruction comes from failing to build python 3.7 for some reason.
> 
> Can you try to download the following release: https://github.com/dora-rs/dora/releases/download/v0.2.4/dora_rs-0.2.4-cp37-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl
> 
> And install it with: pip install dora_rs-0.2.4-cp37-abi3-manylinux_2_17_aarch64.manylinux2014_aarch64.whl`
> 
> You can also try to use a mirror when using pip if your in china such as https://gist.github.com/schnell18/d0ed716917905d2c142a370906cfa32f . That might be the issue.

Thanks xavier, this problem has been solved by reflashing the system.

closed 2023-07-28 by haixuanTao