dora-rs/dora #328: can't run dora -help

Latitude9527, 2023-07-28:
My ubuntu version is aarch 18.04 ,and whern i  install dora-rsv0.2.4,i try to run 'dora --help ',but it shows :
                 bash: /home/agilex/dora_main/dora: Permission denied
          After i run 'chmod +x /home/agilex/dora_main/dora' ,i run 'dora --help' again ,it shows :
                 bash: /home/agilex/dora_main/dora: cannot execute binary file:exec format error
          i had try v0.2.3 and v0.2.4,it shows the same issue;By the way ,my python is 3.7 , my motherboard is Jetson nano.
         Thanks for your reply

Successfully installed dora-rs-0.2.4
agilex@agilex:~/dora-v0.2.4-aarch64-Linux$ PATH=$PATH:$(pwd)
agilex@agilex:~/dora-v0.2.4-aarch64-Linux$ dora --help
bash: /home/agilex/dora_main/dora: cannot execute binary file: Exec format error
agilex@agilex:~/dora-v0.2.4-aarch64-Linux$ dora --version
bash: /home/agilex/dora_main/dora: cannot execute binary file: Exec format error
agilex@agilex:~/dora-v0.2.4-aarch64-Linux$ dora up
bash: /home/agilex/dora_main/dora: cannot execute binary file: Exec format error

meua, 2023-07-28:
HI [Latitude9527](https://github.com/Latitude9527), did you directly unzip these dora binary commands and set them to the environment variable PATH?
https://github.com/dora-rs/dora/releases/download/v0.2.4/dora-v0.2.4-aarch64-Linux.zip

Latitude9527, 2023-07-28:
> 
Thanks ,I have replaced the latest installation package, and now it can run successfully

meua, 2023-07-28:
Nice. make persistent efforts 💪💪💪.

meua, 2023-07-28:
> > 
> 
> Thanks ,I have replaced the latest installation package, and now it can run successfully

Nice. make persistent efforts 💪💪💪.

closed 2023-07-28 by haixuanTao