dora-rs/dora #329: Building on Apple Silicon (M1/M2)

janblumenkamp, 2023-08-02:
**Describe the bug**
I am trying to build dora on Apple Silicon. Building succeeds, but running dora fails due to a missing Python library.

**To Reproduce**
Steps to reproduce the behavior:
I installed cargo and maturin and followed the installation instructions to build from source:
```
git clone https://github.com/dora-rs/dora.git
cd dora
cargo build --release --all 
PATH=$PATH:$(pwd)/target/release
alias dora='dora-cli'

cd apis/python/node
maturin develop --release
cd ../../..
```
Everything runs successfully, but when trying to run `dora`, I get the following error:
```
➜  dora git:(main) dora
dyld[15934]: Library not loaded: '@rpath/libpython3.10.dylib'
  Referenced from: '/Users/jan/Documents/Projects/dora/target/release/dora-cli'
  Reason: tried: '/usr/local/lib/libpython3.10.dylib' (no such file), '/usr/lib/libpython3.10.dylib' (no such file)
[1]    15934 abort      dora-cli
```

**Expected behavior**
I expect dora to run successfully, instead it fails due to a missing python library.

**Environments (please complete the following information):**
 - System info: Apple Macbook Air M2: `Darwin Jans-MacBook-Air-859.local 21.6.0 Darwin Kernel Version 21.6.0: Sat Jun 18 17:07:28 PDT 2022; root:xnu-8020.140.41~1/RELEASE_ARM64_T8110 arm64`
 - Dora version: 89bae20

**Additional context**
N/A

haixuanTao, 2023-08-02:
That is weird indeed,

Can you try building those package separately, as follows:

```bash
cargo build -p dora-coordinator --release
cargo build -p dora-daemon --release
cargo build -p dora-cli --release
```

I don't have an Apple M1 unfortunately.

The error you see comes from, unexpected linking of the python dynamic library.

janblumenkamp, 2023-08-02:
I was able to resolve this by simply deactivating my base conda env!

janblumenkamp, 2023-08-02:
Actually, there seems to be another problem. The examples `c-dataflow`, `rust-dataflow` and `cxx-dataflow` compile and run successfully, but for `python-dataflow` I get the following error when building.

<details>

<summary>Error for building python-dataflow</summary>

```
➜  dora git:(main) cargo run --example python-dataflow
warning: unused variable: `dataflow_descriptor`
  --> binaries/runtime/src/operator/mod.rs:22:5
   |
22 |     dataflow_descriptor: &Descriptor,
   |     ^^^^^^^^^^^^^^^^^^^ help: if this is intentional, prefix it with an underscore: `_dataflow_descriptor`
   |
   = note: `#[warn(unused_variables)]` on by default

warning: `dora-runtime` (lib) generated 1 warning (run `cargo fix --lib -p dora-runtime` to apply 1 suggestion)
    Finished dev [unoptimized + debuginfo] target(s) in 0.28s
     Running `target/debug/examples/python-dataflow`
warning: unused variable: `dataflow_descriptor`
  --> binaries/runtime/src/operator/mod.rs:22:5
   |
22 |     dataflow_descriptor: &Descriptor,
   |     ^^^^^^^^^^^^^^^^^^^ help: if this is intentional, prefix it with an underscore: `_dataflow_descriptor`
   |
   = note: `#[warn(unused_variables)]` on by default

warning: `dora-runtime` (lib) generated 1 warning (run `cargo fix --lib -p dora-runtime` to apply 1 suggestion)
    Finished dev [unoptimized + debuginfo] target(s) in 0.15s
Collecting patchelf
  Using cached patchelf-0.17.2.1.tar.gz (167 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: patchelf
  Building wheel for patchelf (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for patchelf (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [195 lines of output]
      
      
      --------------------------------------------------------------------------------
      -- Trying 'Ninja' generator
      --------------------------------
      ---------------------------
      ----------------------
      -----------------
      ------------
      -------
      --
      CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
        Compatibility with CMake < 3.5 will be removed from a future version of
        CMake.
      
        Update the VERSION argument <min> value or use a ...<max> suffix to tell
        CMake that the project does not need compatibility with older versions.
      
      Not searching for unused variables given on the command line.
      
      CMake Error: CMake was unable to find a build program corresponding to "Ninja".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
      -- Configuring incomplete, errors occurred!
      --
      -------
      ------------
      -----------------
      ----------------------
      ---------------------------
      --------------------------------
[... 154 lines cut ...]
      
      An error occurred while building with CMake.
        Command:
          /opt/homebrew/bin/cmake --build . --target install --config Release --
        Install target:
          install
        Source directory:
          /private/var/folders/rb/ctnwr_1n3cj4_xg5l4l_2_0h0000gn/T/pip-install-ny3bj5l9/patchelf_88047e72e7e74d8eb6e0b9c208f7702c
        Working directory:
          /private/var/folders/rb/ctnwr_1n3cj4_xg5l4l_2_0h0000gn/T/pip-install-ny3bj5l9/patchelf_88047e72e7e74d8eb6e0b9c208f7702c/_skbuild/macosx-12.0-arm64-3.11/cmake-build
      Please check the install target is valid and see CMake's output for more information.
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for patchelf
Failed to build patchelf
ERROR: Could not build wheels for patchelf, which is required to install pyproject.toml-based projects
Error: failed to run python example.

Location:
    examples/python-dataflow/run.rs:34:9
```

</details>

haixuanTao, 2023-08-03:
Thanks for the detailed bug report.
I have removed `patchelf` that should not be necessary anymore. I have also added `macos` in our CI to make sure that the python example run on `macos`.
Do you mind checking out the branch `removing-patchelf` with:
```bash
git pull
git checkout -b removing-patchelf
```
And see if your issue is fixed?

janblumenkamp, 2023-08-03:
Yes, this works! Thank you!

closed 2023-08-03 by janblumenkamp