dora-rs/dora #350: Question. Does the framework provide the concept of Behavior Tree?

besok, 2023-09-05:
Hi there! 
Thanks for the awesome idea and the interesting implementation. 👍 
Great to see new projects in the Rust community dedicated to the sphere of robotics and ml. :smile: 

After a quick poke into the docs, I have not found the API or client to work with Behavior trees like in [ros](https://navigation.ros.org/behavior_trees/index.html)
Have I missed it? If not, do you have plans to implement this part in Dora? 

Currently, I am fiddling around with the behavior trees -> [forester](https://github.com/forester-bt/forester)
and find it engaging to integrate it into Dora by providing the API or in the other way probably. 

Can it be valuable for the project?  

For now, I will make API from the Forester side but If you find it attractive,
I can try to provide some vision into some integration on it as well.

haixuanTao, 2023-09-06:
I just looked at [forester](https://github.com/forester-bt/forester) and it looks great!

I think that a behaviour model is welcomed! We might not be able to set it up as a core feature because, it might add a layer of abstraction that not everyone might need. 

My question would be that we kind of already have a dataflow description that can be used to chain logic so how does [forester](https://github.com/forester-bt/forester) improve the developers abstraction?

besok, 2023-09-06:
Great. Thanks.

> I think that a behaviour model is welcomed! We might not be able to set it up as a core feature because, it might add a layer of abstraction that not everyone might need.

Sure, I see. Obviously, it should be optional. Even, to say more, apparently, just a separate module for the users who are willing to include it. Basically the abstraction above the core broker.

> My question would be that we kind of already have a dataflow description that can be used to chain logic so how does [forester](https://github.com/forester-bt/forester) improve the developers abstraction?

That is a good question. :).  Before all, I need to delve into Dora, to learn the model. 
For now, I can say only common words :) that the [behavior trees](https://forester-bt.github.io/forester/#why-behavior-trees) carry a strong math abstraction and they are well developed and they are fairly popular in robotics (in game design in the first turn)

besok, 2023-09-06:
Great. Thanks.

besok, 2023-09-07:
I will delve it and come back with something :)

closed 2023-09-07 by besok