dora-rs/dora #362: Sending output Arrow array with an empty array cause arrow to panic.

haixuanTao, 2023-10-18:
**Describe the bug**
Sending an output Arrow array with an empty array cause rust arrow to panic.

```bash
thread 'main' panicked at 'Memory pointer from external source (e.g, FFI) is not aligned with the specified scalar type. Before importing buffer through FFI, please make sure the allocation is aligned.', /home/peter/.cargo/registry/src/index.crates.io-6f17d22bba15001f/arrow-buffer-45.0.0/src/buffer/scalar.rs:130:17
```

**To Reproduce**
You can use the python operator example and send an empty vector to a rust node and this would crash the rust node.

This is likely a problem with arrow and not dora directly.

phil-opp, 2023-10-19:
Are you running the `main` branch or the `rust-typed-input` branch? If the latter, what commit are you using?

haixuanTao, 2023-10-19:
The `rust-typed-input`. Sorry I should have probably answered on the PR. I'm using `24ec2f5add5275ea94356c1d8f60a9ecbd421f74`

haixuanTao, 2023-10-19:
Oh wait, let me rebase from origin.

haixuanTao, 2023-10-19:
Ok, so it's still present with the last commit b78f91356183b9cbe1494089f2b8d6f3f129c262 .

You can replicate it by:
```bash
git checkout -b dora-record
cargo run --example python-operator-dataflow
```

phil-opp, 2023-10-19:
Thanks, I'll look into it. I'm currently still working on fixing the C dataflow example, then I'll look into the Python one.

I just fixed one bug that might be related to this in 10c15d50.

haixuanTao, 2023-10-19:
No rush. Thanks!

commit ac7d1740 (2023-10-26): Fix alignment errors when receiving empty arrays

Empty `Vec`s have a dummy pointer that is set to the alignment of the item type. Thus, the pointer for empty `Vec<u8>` data is `0x1`. Using this pointer to construct a `ArrayData` instance can lead to alignment errors if the arrow array is of a type with larger alignment, e.g. float64.

This commit fixes the alignment error by checking for an empty raw buffer and constructing an empty `ArrayData` instance in this case.

Fixes #362
closed 2023-10-26 by phil-opp