dora-rs/dora #379: error in get webcame data

RuPingCen, 2023-11-14:
Hi, haixuanTao. i encountered a problem when i tried to follow the tutorial in the webcam_plot.
https://dora.carsmos.ai/docs/guides/getting-started/webcam_plot

here is the Dora command:
**_dora start dataflow_webcam.yml --attach --hot-reload --name webcam_**

and the terminator return the error messages:
```

3f45130a-a522-48c6-a89e-822dfa18d840
dataflow failed: errors occurred in dataflow 3f45130a-a522-48c6-a89e-822dfa18d840:
- machine ``:
    some nodes failed:
      - plot: 
        3f45130a-a522-48c6-a89e-822dfa18d840/plot failed with exit code 1.
    
        Check logs using: dora logs 3f45130a-a522-48c6-a89e-822dfa18d840 plot
                                
      - web_cam: 
        3f45130a-a522-48c6-a89e-822dfa18d840/web_cam failed with exit code 1.
    
        Check logs using: dora logs 3f45130a-a522-48c6-a89e-822dfa18d840 web_cam
                                
    Location:
        /home/runner/work/dora/dora/binaries/coordinator/src/listener.rs:91:56
```

the **dataflow_webcam.yml** file is :
```

nodes:
  - id: web_cam
    operator:
      python: op_webcam/webcam.py
      inputs:
        tick: dora/timer/millis/100
      outputs:
        - image
  - id: plot
    operator:
      python: op_webcam/plot.py
      inputs:
        image: web_cam/image
```
  

I confused about the error because when I use the python code to show the camera is ok,here is the python code:

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
import cv2
capture = cv2.VideoCapture(0)
while(True):
    ret, frame = capture.read()
    cv2.imshow('frame', frame)
    if cv2.waitKey(1) == ord('q'):
        break 
```

RuPingCen, 2023-11-14:
The log information is :
**dora logs c2979efc-ac20-457b-870a-f027e9151cea webcam**
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
RuntimeError: Dora Runtime raised an error.

Caused by:
   0: main task failed
   1: failed to init an operator
   2: failed to init python operator
   3: SyntaxError: invalid syntax (webcam.py, line 37)

Location:
    binaries/runtime/src/operator/python.rs:30:9
```

haixuanTao, 2023-11-14:
Can you share you're `webcam.py` file?

RuPingCen, 2023-11-14:
The webcam.py is download with the commands

```
wget https://raw.githubusercontent.com/dora-rs/dora/main/examples/python-operator-dataflow/webcam.py
wget https://raw.githubusercontent.com/dora-rs/dora/main/examples/python-operator-dataflow/plot.py
wget https://raw.githubusercontent.com/dora-rs/dora/main/examples/python-operator-dataflow/utils.py
```

[image]

I check the opencv_version, and the terminator output:
4.5.5

haixuanTao, 2023-11-14:
I see. We're using match that has been introduced in python 3.10. Would it be possible to use python 3.10? Otherwise you can replace this match statement with an if-else statement.

RuPingCen, 2023-11-15:
Thanks for your patience, when I use the "python3 --version" to check the python version,  the terminator outputs "Python 3.8.10". So, I am not sure if the python3 version is not right. 

And I replace the match with the if-else but the error still exists. Here is the webcam.py
```

#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import time
from typing import Callable, Optional

import os
import cv2
import numpy as np
import pyarrow as pa

from dora import DoraStatus

CAMERA_WIDTH = 640
CAMERA_HEIGHT = 480
CAMERA_INDEX = int(os.getenv("CAMERA_INDEX", 0))

font = cv2.FONT_HERSHEY_SIMPLEX

class Operator:
    """
    Sending image from webcam to the dataflow
    """

    def __init__(self):
        self.video_capture = cv2.VideoCapture(CAMERA_INDEX)
        self.start_time = time.time()
        self.video_capture.set(cv2.CAP_PROP_FRAME_WIDTH, CAMERA_WIDTH)
        self.video_capture.set(cv2.CAP_PROP_FRAME_HEIGHT, CAMERA_HEIGHT)

    def on_event(
        self,
        dora_event: str,
        send_output: Callable[[str, bytes | pa.Array, Optional[dict]], None],
    ) -> DoraStatus:
        if (dora_event["type"] == "INPUT"):
                ret, frame = self.video_capture.read()
                if ret:
                    frame = cv2.resize(frame, (CAMERA_WIDTH, CAMERA_HEIGHT))

                ## Push an error image in case the camera is not available.
                else:
                    frame = np.zeros((CAMERA_HEIGHT, CAMERA_WIDTH, 3), dtype=np.uint8)
                    cv2.putText(
                        frame,
                        "No Webcam was found at index %d" % (CAMERA_INDEX),
                        (int(30), int(30)),
                        font,
                        0.75,
                        (255, 255, 255),
                        2,
                        1,
                    )

                send_output(
                    "image",
                    pa.array(frame.ravel()),
                    dora_event["metadata"],
                )
        elif (dora_event["type"] =="STOP"):
                print("received stop")
        else:
                print("received unexpected event:", other)

        if time.time() - self.start_time < 20:
            return DoraStatus.CONTINUE
        else:
            return DoraStatus.STOP

    def __del__(self):
        self.video_capture.release()
```
[image]

And the log info is :
[image]

haixuanTao, 2023-11-15:
I will remove type hinting and the match statement from the exemple for better legacy python support. 

Thanks for raising the issue. Where you able to make the webcam run?

RuPingCen, 2023-11-16:
Thanks for your help,  this problem has solved with the 
```
 send_output: Callable[[str, bytes | pa.Array, Optional[dict]], None],
    ) -> DoraStatus:
```
to be 
```
 send_output: Callable[[str, bytes or pa.Array, Optional[dict]], None],
    ) -> DoraStatus:
```

also, the **match** is replaced with **if-else** .

RuPingCen, 2023-11-16:
But,I found that using dora webcam to plot the USB camera will cause obvious delay, but using opencv display does not. 
```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
import cv2
capture = cv2.VideoCapture(0)
while(True):
    ret, frame = capture.read()
    cv2.imshow('frame', frame)
    if cv2.waitKey(1) == ord('q'):
        break 
```
Have you ever encountered such a phenomenon?

haixuanTao, 2023-11-16:
So, you have to check your tick rate:
```yaml
        tick: dora/timer/millis/100
```

You can change it to something like 

```yaml
        tick: dora/timer/millis/50
```

And you will have a smoother video stream normally.

RuPingCen, 2023-11-16:
Yes, it is work, Thank you very much.

PR #382 (2023-11-16): Support legacy python
This PR removes:
- type hinting as it is a source of errors for legacy python version
- removes `match` as it is only supported since python `3.10` 

that is unfortunately that legacy python is still so common

 FIx: #379
closed 2023-11-16 by RuPingCen