dora-rs/dora #393: How can I use dora for model training?

npuichigo, 2023-12-15:
Fantastic work first!

But I cannot find any example on how to use dora for general model training, saying I would like to use dora as a data pipeline for training in PyTorch, for example. I think it's good to provide an toy example like mnist

haixuanTao, 2023-12-18:
Absolutely! Are you more interested in real-time training? Or you prefer do off-line training within you're regular environment? The ladder should be easier to do using dora-record, as the exported format is arrow so that you can read it with pandas and then do everything with pandas, numpy, torch, tensorflow...

npuichigo, 2023-12-18:
I'm wondering if dora is the same as ray or maybe there's something special?

haixuanTao, 2023-12-18:
Are you referring to: https://github.com/ray-project/ray 
In which case, I think they focus on giving a full-stack pipeline for AI Model training and serving. We're more focused on producing real-time AI-based application that can typically run on a car like dora-drives or a small robot like dora-robomaster. 

One of the challenge we aim at tackling is things like integrating sensors, integrating controls, all of this in an environment that is easy to dev with, with feature like hot-reloading, opentelemetry tracing, metrics and logging, and much more :) 

We can do training through dora-record to export the data collected in arrow format. And you can then use your most loved software stack to do the training. This can be ray!