dora-rs/dora #401: flake8 5.0.4 requires importlib-metadata<4.3,>=1.1.0; python_version < "3.8", but you have importlib-metadata 6.0.0 which is incompatible.

caixxuan, 2023-12-22:
**Describe the bug**
something wrong when "pip install -r install_requirements.txt & pip install -r requirements.txt":
A clear and concise description of what the bug is.
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
opentelemetry-api 1.22.0 requires importlib-metadata<7.0,>=6.0, but you have importlib-metadata 4.2.0 which is incompatible.
Successfully installed einops-0.3.2 importlib-metadata-4.2.0
(dora3.7) cai@cai:~/Downloads/dora-drives$ pip install importlib-metadata==6.0.0
Collecting importlib-metadata==6.0.0
  Using cached importlib_metadata-6.0.0-py3-none-any.whl (21 kB)
Requirement already satisfied: zipp>=0.5 in /home/cai/anaconda3/envs/dora3.7/lib/python3.7/site-packages (from importlib-metadata==6.0.0) (3.15.0)
Requirement already satisfied: typing-extensions>=3.6.4 in /home/cai/anaconda3/envs/dora3.7/lib/python3.7/site-packages (from importlib-metadata==6.0.0) (4.3.0)
Installing collected packages: importlib-metadata
  Attempting uninstall: importlib-metadata
    Found existing installation: importlib-metadata 4.2.0
    Uninstalling importlib-metadata-4.2.0:
      Successfully uninstalled importlib-metadata-4.2.0
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
flake8 5.0.4 requires importlib-metadata<4.3,>=1.1.0; python_version < "3.8", but you have importlib-metadata 6.0.0 which is incompatible.
Successfully installed importlib-metadata-6.0.0

**To Reproduce**
Steps to reproduce the behavior:
1. Dora start daemon: `dora up` 
2. Start a new dataflow: `dora start dataflow.yaml`
3. Stop dataflow: `dora stop`
4. Destroy dataflow: `dora destroy`

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or Video**
If applicable, add screenshots to help explain your problem.

**Environments (please complete the following information):**
 - System info: [use `uname --all` on Linux]
 - Dora version: [use `dora --version` and `pip show dora-rs`]

**Additional context**
Add any other context about the problem here.

caixxuan, 2023-12-22:
[image]
if i force to run, something wrong happen

caixxuan, 2023-12-22:
[image]
moreover, the carla seems have something wrong when starting up

haixuanTao, 2023-12-22:
For:
- the dependency, I have to double check the latest pip requirements. But It's possible that newer version of opentelemetry are incompatible with dora-drives.
- For the error, I think there is something wrong within carla_gps_op and the `opendrive` input.
- The carla environment seems ok for me but you're seems to be running a version without GPU. Make sure that nvidia-docker and your CUDA environment are well installed