dora-rs/dora #402: The header file 'operator_typers.h' does not provide the specific definitions for the 'Input' and 'Output' structures.

ShoreFlower, 2023-12-23:
The header file 'operator_typers.h' does not provide the specific definitions for the 'Input' and 'Output' structures. 

/** <No documentation available> */
typedef struct Input Input_t;

/** <No documentation available> */
typedef struct Output Output_t;

ShoreFlower, 2023-12-23:
I think it should be this:

/** <No documentation available> */
typedef struct Input{
    Vec_uint8_t id;
    Vec_uint8_t data;
} Input_t;

/** <No documentation available> */
typedef struct Output{
    Vec_uint8_t id;
    Vec_uint8_t data;
} Output_t;

phil-opp, 2024-01-14:
@ShoreFlower To access the `id` and `data` of inputs, you can use the `dora_read_input_id` and `dora_read_data` functions.