dora-rs/dora #403: Fail To Compile C-Dataflow Example

RuPingCen, 2023-12-24:
**Describe the bug**
I tried to compile C-Dataflow example in the GitHub repository,  https://github.com/dora-rs/dora/tree/main/examples/c-dataflow , in the step  "Compile the dora-node-api-c crate into a static library", 

`cargo build -p dora-node-api-c --release`

the compiler prompted that a file was missing：

**_error: could not find `Cargo.toml` in `/home/dora/dora_project/examples/c-dataflow` or any parent directory_**

But this file was not found in the GitHub repository.

[image]

haixuanTao, 2023-12-28:
You should do this inside of the Dora repository:

```bash
git clone git@github.com:dora-rs/dora.git
cd dora 
cargo build -p dora-node-api-c --release
```

You can then use the shared library as you like

RuPingCen, 2024-01-03:
Thank you for your answer. The compilation issue has been resolved.

RuPingCen, 2024-01-03:
And,  Another problem is troubling me. I created two C++nodes, the node A publishes a count, and node B receives and shows data published by node A. The length of the data I read using the function   **_read_dora_input_data(event, &data_ptr, &data_len);_**  has always been 0, but the ID string is right.

The code for node A as follows:

```
 extern "C"
{
#include "/home/dora/dora_project/dora/apis/c/node/node_api.h"   
#include "/home/dora/dora_project/dora/apis/c/operator/operator_api.h"
#include "/home/dora/dora_project/dora/apis/c/operator/operator_types.h"
}

#include <iostream>
#include <vector>

  
bool to_exit_process = false;
 
int run(void *dora_context)
{
    unsigned char counter = 0;

    //for (int i = 0; i < 20; i++)
    to_exit_process = false;
    while(!to_exit_process)
    {
        void *event = dora_next_event(dora_context);
        if (event == NULL)
        {
            printf("[c node] ERROR: unexpected end of event\n");
            return -1;
        }

        enum DoraEventType ty = read_dora_event_type(event);

        if (ty == DoraEventType_Input)
        {
            counter += 1;
             std::string out_id = "pointcloud";
            int resultend=dora_send_output(dora_context, &out_id[0], out_id.length(), (char *)&counter, 1);
            std::cout
                << "dora_send_output: out_id "<<out_id<< "  out_data: "<<(unsigned int)counter<<std::endl;
             if (resultend != 0)
            {
                std::cerr << "failed to send output" << std::endl;
                return 1;
            }
        }
        else if (ty == DoraEventType_Stop)
        {
            printf("[c node] received stop event\n");
        }
        else
        {
            printf("[c node] received unexpected event: %d\n", ty);
        }

        free_dora_event(event);
    }
    return 0;
}

int main()
{
    auto dora_context = init_dora_context_from_env();
    auto ret = run(dora_context);
    free_dora_context(dora_context);
    
    to_exit_process = true;
 
    std::cout << "exit rslidar driver ..." << std::endl;
    return ret;
}
```

The code of node B as follows:
```
// 配合rslidar 在dora中的驱动节点，接收rslidar dora驱动节点的数据 并进行显示打印，确保驱动节点正常工作

extern "C"
{
#include "/home/dora/dora_project/dora/apis/c/node/node_api.h"   
#include "/home/dora/dora_project/dora/apis/c/operator/operator_api.h"
#include "/home/dora/dora_project/dora/apis/c/operator/operator_types.h"
}

#include <iostream>
#include <vector>
// rs lidar driver
#include <rs_driver/api/lidar_driver.hpp>

#ifdef ENABLE_PCL_POINTCLOUD
#include <rs_driver/msg/pcl_point_cloud_msg.hpp>
#else
#include <rs_driver/msg/point_cloud_msg.hpp>
#endif
typedef PointXYZI PointT;
typedef PointCloudT<PointT> PointCloudMsg;

using namespace robosense::lidar;

SyncQueue<std::shared_ptr<PointCloudMsg>> free_cloud_queue;
SyncQueue<std::shared_ptr<PointCloudMsg>> stuffed_cloud_queue;
  
  
bool to_exit_process = false;
 
int run(void *dora_context)
{
    unsigned char counter = 0;
    //for (int i = 0; i < 20; i++)
    to_exit_process = false;
    while(!to_exit_process)
    {
        void *event = dora_next_event(dora_context);
        if (event == NULL)
        {
            printf("[c node] ERROR: unexpected end of event\n");
            return -1;
        }

        enum DoraEventType ty = read_dora_event_type(event);

        if (ty == DoraEventType_Input)
        {
            counter += 1;

            char *id_ptr;
            size_t id_len;
            read_dora_input_id(event, &id_ptr, &id_len);
            std::string id(id_ptr, id_len);
            char *data_ptr;
            size_t data_len = 1;
            read_dora_input_data(event, &data_ptr, &data_len);

            std::cout<< "C++ Node (C-API) received input: " << id <<" id_len:"<<id_len
                    <<" data_len:"<<data_len<<std::endl;
            // --------------------- 
            std::vector<unsigned char> data;
            for (size_t i = 0; i < data_len; i++)
            {
                data.push_back(*(data_ptr + i));
            }
            std::cout
                << "Debug: Received input " << " (counter: " << (unsigned int)counter <<"  data.size():   "<< data.size()<< ") data: [";
            for (unsigned char &v : data)
            {
                std::cout << (unsigned int)v << ", ";
            }
            std::cout << "]" << std::endl<< std::endl<< std::endl;
  
        }
        else if (ty == DoraEventType_Stop)
        {
            to_exit_process = true;
            printf("[c node] received stop event\n");
        }
        else
        {
            to_exit_process = true;
            printf("[c node] received unexpected event: %d\n", ty);
        }

        free_dora_event(event);
    }
    return 0;
}

int main()
{
    std::cout << "Debug rslidar driver in dora: " << std::endl;
 
    auto dora_context = init_dora_context_from_env();
    auto ret = run(dora_context);
    free_dora_context(dora_context);

    
    to_exit_process = true;
    
    std::cout << "exit Debug rslidar ..." << std::endl;
    return ret;
}

```

The yaml file as follow:
[image]

 And I found that the length of the data returned by the function **_read_dora_input_data(event, &data_ptr, &data_len)_**  is 0 

[image]

haixuanTao, 2024-01-03:
Indeed. Thanks for reporting this error. The PR #406 should fix your issue. Could you try it out with:
```bash
git checkout fix-cxx-node-input
```

PR #406 (2024-01-04): Fix read error in C++ node input
Thwy were an error of typecasting within the C++ node input.

This fixes #403

I have not adressed all input type as I think a solution based on arrow FFI would be better.
commit 06a7d0bf (2024-01-04): Fix read error in C++ node input

Thwy were an error of typecasting within the C++ node input.

This fixes #403

I have not adressed all input type as I think a solution based on arrow FFI would be better.
closed 2024-01-04 by phil-opp