dora-rs/dora #404: two questions: 1. how to set the destination for ego vehicle? 2. how to print the variable?

caixxuan, 2023-12-25:
1. how to set the destination for ego vehicle? 
I did not find the file to set  of destination carla ego vehicle. how to set it?
2. how to print the variable?
I want to monitor some variables when running simulation, but it seems did not work using 'print(xx)'. are there some other ways to realize?

caixxuan, 2023-12-27:
[image]
if set the NUM_WAYPOINTS_AHEAD from 120 to 145，something wrong happened, the ego vehicle cannot work properly, how to fix?

caixxuan, 2023-12-27:
> In order to change the goal destination you should look at the carla_gps_op.py and change it there.
> 
> in order to print anything you should us: dora logs <DATAFLOW_ID>
> 
> To see it printing there.
> 
> check the flush=true option in print to see in real time

sorry, I am not clear: 'dora logs <DATAFLOW_ID>' should write to the python script? and where to find '<DATAFLOW_ID>' ? flush=true where to set?

caixxuan, 2023-12-27:
> sorry, I am not clear: 'dora logs <DATAFLOW_ID>' should write to the python script? and where to find '<DATAFLOW_ID>' ? flush=true where to set?

is there any example code?

haixuanTao, 2023-12-28:
> [image]
> 
> if set the NUM_WAYPOINTS_AHEAD from 120 to 145，something wrong happened, the ego vehicle cannot work properly, how to fix?

I wouldn't recommend changing the number of waypoint ahead. You should probably revert back.

However, if you really want to you should change make sure that every message using the waypoints are reshaping the data in the right shape with this newly updated number of waypoints.

haixuanTao, 2023-12-28:
> sorry, I am not clear: 'dora logs <DATAFLOW_ID>' should write to the python script? and where to find '<DATAFLOW_ID>' ? flush=true where to set?

You should type this in your terminal:

```bash
Dora logs c95d118b-cded-4531-a0e4-cd85b7c3916c carla_gps_op
```

In your Python script make sure to use:

```
print("test", flush=True)
```

caixxuan, 2024-01-02:
It seems only something wrong happen, the print can work?
"description should only be set when status_code is set to StatusCode.ERROR"

caixxuan, 2024-01-02:
> > > In order to change the goal destination you should look at the carla_gps_op.py and change it there.
> > 
> > 
> > > 
> > 
> > 
> > > in order to print anything you should us: dora logs <DATAFLOW_ID>
> > 
> > 
> > > 
> > 
> > 
> > > To see it printing there.
> > 
> > 
> > > 
> > 
> > 
> > > check the flush=true option in print to see in real time
> > 
> > 
> > sorry, I am not clear: 'dora logs <DATAFLOW_ID>' should write to the python script? and where to find '<DATAFLOW_ID>' ? flush=true where to set?
> 
> You should type this in your terminal:
> 
> ```shell
> Dora logs c95d118b-cded-4531-a0e4-cd85b7c3916c carla_gps_op
> ```
> 
> In your Python script make sure to use:
> 
> ```python
> print("test", flush=True)
> ```

it seems only work when fail:
description should only be set when status_code is set to StatusCode.ERROR

caixxuan, 2024-01-02:
> You should type this in your terminal:
>
> ```shell
> Dora logs c95d118b-cded-4531-a0e4-cd85b7c3916c carla_gps_op
> ```
>
> In your Python script make sure to use:
>
> ```python
> print("test", flush=True)
> ```

And the print is not executed in real time

haixuanTao, 2024-01-02:
So the print of logs is not in real time and you will have to open and close it to take into account the latest logs. We're working on a better logging system but as of now, this is how we do logging.

haixuanTao, 2024-04-21:
You can now check the logs in the `out` folder.

closed 2024-04-21 by haixuanTao