dora-rs/dora #408: When Dora node exit unexpectedly,there will be some resources in the kernel that cannot be released.

RuPingCen, 2024-01-09:
**Describe the bug**
We found a phenomenon,but i am not sure if it is a dora bug.

We created 3 nodes A, B, C in Dora. Node A is responsible for reading IMU measurements from the serial port(/dev/ttyUSB0), such as xsens mti-30. Node B receives the data from node A and prints the data to confirm that node A can read the IMU data normally. Node C is a node under development. We hope to use node C to receive the data of node A and forward it to ROS2.
Node A and node B can run normally without reporting errors.
When we were debugging node C, we found that if node C exits abnormally, it will cause dora kernel sotp (I am not sure whether the description of sotp is correct), that is, we need to start it again with the dora up command before we can continue to use dora start.

The phenomenon is like this：

Once node C fails, the dora kernel stops, node A cannot be started again, and the node will prompt that the serial port is always occupied. In other words, the serial port may not be released after dora exits abnormally.
 
 This phenomenon has occurred several times, and we currently can only solve this problem by restarting the computer.

haixuanTao, 2024-01-09:
Do you mind sharing the errors logs that you get so that we can help you?

To kill a running node, you can use:
```bash
pkill -f 'name of your command'
```
It's an open issue #184  that killing node that leave for too long.

Sorry for the inconvenience.

haixuanTao, 2024-08-31:
I think that this has been solved with the addition of grace duration

closed 2024-08-31 by haixuanTao