dora-rs/dora #427: Error when modifying dora-rs node

WindsAtr, 2024-02-15:
I wanted to add the distribution_location_op/obstacles data flow in carla_gps_op.py, so I modified the node id in the oasis_agent_planning.yaml file: carla_gps_op, but the carla_gps_op.py file did not obtain the distribution_location_op/obstacles data.
This is the part I modified：
  - id: carla_gps_op
    operator:
      python: ../../carla/carla_gps_op.py
      outputs:
        - gps_waypoints
      inputs:
        opendrive: oasis_agent/opendrive
        objective_waypoints: oasis_agent/objective_waypoints
        position: oasis_agent/position
        obstacles: obstacle_location_op/obstacles
[image]

WindsAtr, 2024-02-15:
can anyone help me

haixuanTao, 2024-02-15:
The way you modified the dataflow seems right. Are you sure that there is obstacles in front?

haixuanTao, 2024-02-19:
This has been solved in the discord

closed 2024-02-19 by haixuanTao