dora-rs/dora #437: Use new `Bound` API of `pyO3`

phil-opp, 2024-03-15:
We use the [`pyO3`](https://github.com/pyo3/pyo3) crate for implementing Python nodes and operators. The upcoming `v0.21` release will feature a new "Bound" API, which enables better control over the lifetimes of GIL-references, while also improving performance. It also solve https://github.com/PyO3/pyo3/issues/1056, which we encountered in dora.

Right now `v0.21` is still [in beta](https://github.com/PyO3/pyo3/pull/3944), but we should try to migrate dora to the new [`Bound` ](https://docs.rs/pyo3/0.21.0-beta.0/pyo3/prelude/struct.Bound.html) API once the stable is out.

(Maybe it will even allow us to bring back runtime nodes with multiple Python operators?)

haixuanTao, 2024-03-15:
I think this could solve one of the issues with drop tokens reporting at the end of running dataflows.

haixuanTao, 2024-03-27:
https://github.com/PyO3/pyo3/releases/tag/v0.21.0 

`0.21` has been released!

closed 2024-06-08 by haixuanTao