dora-rs/dora #438: Does dora currently provide a demo of multi-device cross-network communication?

baoliete, 2024-03-16:


heyong4725, 2024-03-18:
Originally Dora uses Zenoh as communication middleware, which can enable Dora to support multi-device cross-network communications. Most recent use cases are focused on local device, such that shared memory is the main communication middleware. With ROS2 extension, the DDS can be used for multi-device communication (but not for cross network).

zhuyifengzju, 2024-04-20:
Hi @heyong4725, do you mean that if we want to use dora for multi-device communication (but assuming within a network), we will need to use the ROS 2 extension? Or is it possible to directly use dora-daemon that bind to the same ip address?

haixuanTao, 2024-04-21:
@zhuyifengzju , we're currently working on the multi-device communication, but it is not there yet see: #459 .
We've been primarily focused on one machine and was thinking of bringing very pleasant developers experience within multi-devices. How do we deal with syncing multi-device file system and development. That is why it took a bit of time but is definetely on out todo list.

Obviously without ros2

zhuyifengzju, 2024-04-21:
Got it. Look forward to its release!