dora-rs/dora #444: 类型 char* 转化为 ::rust::Vec<::std::uint8_t>  的问题。The problem of converting type char* to ::rust::Vec<::std::uint8_t>.

ShoreFlower, 2024-03-28:
在应用ROS消息类型的时候存在字节转化问题，临时使用的`for`循环转化,效率较低，请问有比较高效的方式么？比如直接copy指针，C++提供的copy无法执行。

When using ROS message types, there is a byte conversion issue. The temporary solution is to use a for loop for conversion, which is inefficient. Is there a more efficient way, such as directly copying pointers? The copy provided by C++ does not work.
### bug code
```
  for (int i = 16; i<output_data_len; i++) {
            PointCloud2.data.push_back(static_cast<uint8_t>(output_data[i]));
        }
```

### all code
```

#include "dora-node-api.h"
#include "dora-ros2-bindings.h"

#include <iostream>
#include <vector>
#include <random>

// rs lidar driver
#include <rs_driver/api/lidar_driver.hpp>

#ifdef ENABLE_PCL_POINTCLOUD
#include <rs_driver/msg/pcl_point_cloud_msg.hpp>
#else
#include <rs_driver/msg/point_cloud_msg.hpp>
#endif
typedef PointXYZI PointT;                       // x,y,z, intensity;
                                                // intensity: ji guang fan she qiang du  
                                                // this is a point in the point cloud

/// @brief PointCloudMsg
//  typedef std::vector<PointT> VectorT
//  uint32_t height = 0
//  uint32_t width = 0
//  bool is_dense = false
//  double timestamp = 0
//  uint32_t seq = 0  //sequence number of message
//  std::string frame_id = ""  
//  vectorT points
typedef PointCloudT<PointT> PointCloudMsg;       

using namespace robosense::lidar;

SyncQueue<std::shared_ptr<PointCloudMsg>> free_cloud_queue;
SyncQueue<std::shared_ptr<PointCloudMsg>> stuffed_cloud_queue;

//
// @brief point cloud callback function. The caller should register it to the lidar driver.
//        Via this fucntion, the driver gets an free/unused point cloud message from the caller.
// @param msg  The free/unused point cloud message.
//
std::shared_ptr<PointCloudMsg> driverGetPointCloudFromCallerCallback(void)//从free队列里面取
{
  // Note: This callback function runs in the packet-parsing/point-cloud-constructing thread of the driver, 
  //       so please DO NOT do time-consuming task here.
  std::shared_ptr<PointCloudMsg> msg = free_cloud_queue.pop();
  if (msg.get() != NULL)
  {
    return msg;
  }

  return std::make_shared<PointCloudMsg>();
}

//
// @brief point cloud callback function. The caller should register it to the lidar driver.
//        Via this function, the driver gets/returns a stuffed point cloud message to the caller. 
// @param msg  The stuffed point cloud message.
[... 223 lines cut ...]
        // std::copy(output_data+16, output_data + output_data_len, PointCloud2.data);
        // std::vector 
        // std::copy(output_data.begin(), output_data.end(), PointCloud2.data);
        // std::memcpy(PointCloud2.data, output_data, output_data_len);

        for (int i = 16; i<output_data_len; i++) {
            PointCloud2.data.push_back(static_cast<uint8_t>(output_data[i]));
        }
        
        delete []output_data;
        output_data = NULL;
        vel_publisher->publish(PointCloud2);
    }
    to_exit_process = true;
    driver.stop();
       

   

    std::cout << "exit rslidar driver ..." << std::endl;

    return 0;
}

```

haixuanTao, 2024-04-01:
I'm not an expert of C++, but maybe try adding `.data()` at the end: 
```c++
std::memcpy(PointCloud2.data.data(), output_data, output_data_len);
```

From memory alone, I think that `PointCloud2.data` is a standard vector in C++

https://chat.openai.com/c/c42af36d-9385-4db5-a168-c1337d570335

p.s: note that you can use ```c++ for syntax highlighting

phil-opp, 2024-04-03:
See https://cxx.rs/binding/vec.html for the type definition of `rust::Vec<T>`. While it's similar to a `std::vector`, it's a different type without any direction conversion. Unfortunately, I don't see any more efficient way to push multiple values either.

> ```c++
> std::memcpy(PointCloud2.data.data(), output_data, output_data_len);
> ```

This is unsafe and will lead to undefined behavior if `output_data_len` is greater than `PointCloud2.data.size()`.

---

A possible solution could be to generate two versions of each message type: one based on Rust types and one based on C++ types. The version based on C++ types would then contain normal `std::vector` fields instead of `rust::Vec`. Subscribe methods would only support the version based on Rust-types, but publish methods could support both types. The drawback is that it duplicates all message types, which could be confusing. What do you think about this approach @ShoreFlower?

ShoreFlower, 2024-04-03:
I don't think this is a good approach.

---- 回复的原邮件 ----
| 发件人 | Philipp ***@***.***> |
| 日期 | 2024年04月03日 19:30 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [dora-rs/dora] 类型 char* 转化为 ::rust::Vec<::std::uint8_t>  的问题。The problem of converting type char* to ::rust::Vec<::std::uint8_t>. (Issue #444) |

See https://cxx.rs/binding/vec.html for the type definition of rust::Vec<T>. While it's similar to a std::vector, it's a different type without any direction conversion. Unfortunately, I don't see any more efficient way to push multiple values either.

std::memcpy(PointCloud2.data.data(), output_data, output_data_len);

This is unsafe and will lead to undefined behavior if output_data_len is greater than PointCloud2.data.size().

A possible solution could be to generate two versions of each message type: one based on Rust types and one based on C++ types. The version based on C++ types would then contain normal std::vector fields instead of rust::Vec. Subscribe methods would only support the version based on Rust-types, but publish methods could support both types. The drawback is that it duplicates all message types, which could be confusing. What do you think about this approach @ShoreFlower?

phil-opp, 2024-04-03:
To avoid the `for` loop, you should be able to use [`back_inserter`](https://en.cppreference.com/w/cpp/iterator/back_inserter) together with `std::copy`.

Doing a pointer copy would require a [public `set_len` method](https://github.com/dtolnay/cxx/issues/990), which is being discussed upstream. However, apparently such a function isn't even available for the standard C++ `vector` type.

If you're concerned about efficiency, you should try doing a benchmark with link-time optimization enabled. I can imagine that the compiler/linker optimizes the `for` loop and `push_back` calls to some quite efficient code.

phil-opp, 2024-04-03:
To make the `push_back` operation cheap you can call `reserve` with the required capacity beforehand. Then no reallocation is required anymore.