dora-rs/dora #446: How to use CMake to build  c++-ros2-dataflow

bobd988, 2024-03-28:
We are trying to start dora integration with examples/c++-ros2-dataflow
There are some C++ existing system built with cmake while dora c++-ros2-dataflow  is built with cargo.  Can this c++-ros2-dataflow example provide a cmake example so that dora modules can be easily integrated to cmake system?

There is a similar one in examples/cmake-dataflow which may be used as a base , however 
when running the below from a Ubuntu 20.04 with Cmake 3.28  

cd build && cmake ..
make install

I got  error: failed to run custom build command for `dora-node-api-cxx v0.3.2 (/home/demo/Public/github_dora/dora/examples/cmake-dataflow/build/dora/src/Dora/apis/c++/node)`

It will be great if this cmake example cmake build issue can be resolved. And then apply to c++-ros2-dataflow exmaple ? Please provide instructions how to build and run , for example the current command to run is 

cargo run --example cxx-ros2-dataflow --features ros2-examples

After applied to cmake, do we still run with this or different command ?

phil-opp, 2024-03-30:
I'm not familiar with cmake, so I'm not sure if I can help you with the details, but I try to take a look.

Maybe @XxChang can help?

XxChang, 2024-03-30:
Please run it with `cmake -DDORA_ROOT_DIR=<path-to-dora>`, see [run.rs](https://github.com/dora-rs/dora/blob/main/examples/cmake-dataflow/run.rs) for more details.

LoL. My negligence.

The case of out-of-tree compile fails because the `target_dir()` in [build.rs](https://github.com/dora-rs/dora/blob/main/apis/c%2B%2B/node/build.rs) was changed in last version.  But it is ok with `cargo run --example cmake-dataflow`.

To fix these bugs, please modify [DoraTargets.cmake](https://github.com/dora-rs/dora/blob/main/examples/cmake-dataflow/DoraTargets.cmake)
replace
```
        BUILD_COMMAND
            cargo build
            --package dora-node-api-c
            --target-dir ${CMAKE_CURRENT_BINARY_DIR}
            &&
            cargo build
            --package dora-operator-api-c
            --target-dir ${CMAKE_CURRENT_BINARY_DIR}
            &&
            cargo build
            --package dora-node-api-cxx
            --target-dir ${CMAKE_CURRENT_BINARY_DIR}
            &&
            cargo build
            --package dora-operator-api-cxx
            --target-dir ${CMAKE_CURRENT_BINARY_DIR}
```
with
```
        BUILD_COMMAND
            cargo build
            --package dora-node-api-c
            --target-dir ${CMAKE_CURRENT_BINARY_DIR}/src/Dora/target
            &&
            cargo build
            --package dora-operator-api-c
            --target-dir ${CMAKE_CURRENT_BINARY_DIR}/src/Dora/target
            &&
            cargo build
            --package dora-node-api-cxx
            --target-dir ${CMAKE_CURRENT_BINARY_DIR}/src/Dora/target
            &&
            cargo build
            --package dora-operator-api-cxx
            --target-dir ${CMAKE_CURRENT_BINARY_DIR}/src/Dora/target
```
and `set(dora_link_dirs ${CMAKE_CURRENT_BINARY_DIR}/debug)` -> `set(dora_link_dirs ${CMAKE_CURRENT_BINARY_DIR}/src/Dora/target/debug)`

how to build with a existing cmake project? May be you can learn more from this [repo](https://github.com/XxChang/dora_simple_vo)

phil-opp, 2024-03-30:
Thanks a lot!

> To fix these bugs, please modify [DoraTargets.cmake](https://github.com/dora-rs/dora/blob/main/examples/cmake-dataflow/DoraTargets.cmake)
> replace

So we should perform these changes in the repo?

XxChang, 2024-03-30:
There is no problem with `cargo run --example cmake-dataflow`, so it is ok with test scripts.

but when people want to do it on a separate project, it will fail. So please perform these changes. Thanks.

bobd988, 2024-04-01:
@XxChang  with current latest code from main , to add  above changes in DoraTargets.cmake, When I run `cmake -DDORA_ROOT_DIR=/home/demo/Public/github_dora/dora`, i got  

>

>  "/home/demo/Public/github_dora/dora" does not appear to contain CMakeLists.txt ,  

 then I change to run  ` cmake -DDORA_ROOT_DIR=/home/demo/Public/github_dora/dora/examples/cmake-dataflow` , i got    

> execute_process failed command indexes:
> 
>     1: "Child return code: 101"
> 
> Call Stack (most recent call first):
>   _deps/corrosion-src/cmake/CorrosionGenerator.cmake:217 (_cargo_metadata)
>   _deps/corrosion-src/cmake/Corrosion.cmake:1051 (_generator_add_cargo_targets)
>   DoraTargets.cmake:19 (corrosion_import_crate)
>   CMakeLists.txt:7 (include)

It may also caused by my Ubuntu environment. Let me verify this with another ubuntu 20.04

XxChang, 2024-04-01:
You should run `cmake -DDORA_ROOT_DIR=<path-to-dora>` under the  `cmake-dataflow` directory 

or

```
cd <path-to-cmake-dataflow-directory>
mkdir build
cd build
cmake -DDORA_ROOT_DIR=<path-to-dora> ..
make
```

phil-opp, 2024-04-03:
@XxChang 

> but when people want to do it on a separate project, it will fail. So please perform these changes. Thanks.

I opened a PR at #453

bobd988, 2024-04-06:
I got different error when running `cargo run --example cmake-dataflow `under ubuntu22.04. Latest dora code from main    

> 
>  Finished dev [unoptimized + debuginfo] target(s) in 11.70s
>      Running `target/debug/examples/cmake-dataflow`
> -- Rust Toolchain: 1.76-x86_64-unknown-linux-gnu
> -- Rust Target: x86_64-unknown-linux-gnu
> -- Found Rust: /home/demo/.rustup/toolchains/1.76-x86_64-unknown-linux-gnu/bin/rustc (found version "1.76.0") 
> -- Using Corrosion as a subdirectory
> -- Rust Toolchain: 1.76-x86_64-unknown-linux-gnu
> -- Rust Target: x86_64-unknown-linux-gnu
> -- Found Rust: /home/demo/.rustup/toolchains/1.76-x86_64-unknown-linux-gnu/bin/rustc (found suitable version "1.76.0", minimum required is "1.72") 
> -- Configuring done (0.4s)
> -- Generating done (0.0s)
> -- Build files have been written to: /home/demo/Public/dora/examples/cmake-dataflow/build
> [  0%] Built target cargo-prebuild_dora-node-api-c
>     Finished dev [unoptimized + debuginfo] target(s) in 0.15s
> Copying byproducts `libdora_node_api_c.a` to /home/demo/Public/dora/examples/cmake-dataflow/build
> [  0%] Built target _cargo-build_dora-node-api-c
> [  0%] Built target cargo-build_dora-node-api-c
> [  0%] Built target cargo-prebuild_dora-operator-api-c
>     Finished dev [unoptimized + debuginfo] target(s) in 0.13s
> Copying byproducts `libdora_operator_api_c.a` to /home/demo/Public/dora/examples/cmake-dataflow/build
> [  0%] Built target _cargo-build_dora-operator-api-c
> [  0%] Built target cargo-build_dora-operator-api-c
> [  0%] Built target cargo-prebuild_dora-node-api-cxx
>     Finished dev [unoptimized + debuginfo] target(s) in 0.11s
> Copying byproducts `libdora_node_api_cxx.a` to /home/demo/Public/dora/examples/cmake-dataflow/build
> [  0%] Built target _cargo-build_dora-node-api-cxx
> [  0%] Built target cargo-build_dora-node-api-cxx
> [  0%] Built target cargo-prebuild_dora-operator-api-cxx
>     Finished dev [unoptimized + debuginfo] target(s) in 0.08s
> Copying byproducts `libdora_operator_api_cxx.a` to /home/demo/Public/dora/examples/cmake-dataflow/build
> [  0%] Built target _cargo-build_dora-operator-api-cxx
> [  0%] Built target cargo-build_dora-operator-api-cxx
> [  8%] Built target Dora_c
> [ 25%] Built target node_c_api
> [ 33%] Generating node_bridge.cc, include/cxx, operator_bridge.cc
> cp: cannot stat 'target/cxxbridge/dora-node-api-cxx/src/lib.rs.cc': No such file or directory
> gmake[2]: *** [CMakeFiles/Dora_cxx.dir/build.make:75: node_bridge.cc] Error 1
> gmake[1]: *** [CMakeFiles/Makefile2:630: CMakeFiles/Dora_cxx.dir/all] Error 2
> gmake: *** [Makefile:136: all] Error 2
> Error: failed to build a cmake-generated project binary tree
>

XxChang, 2024-04-07:
see PR https://github.com/dora-rs/dora/pull/453

XxChang, 2024-04-07:
> I got different error when running `cargo run --example cmake-dataflow `under ubuntu22.04. Latest dora code from main

fix it on https://github.com/dora-rs/dora/pull/460

closed 2024-04-21 by haixuanTao