dora-rs/dora #447: latest main compile error

bobd988, 2024-03-29:
The previous version worked fine but with  the latest from main branch in same ubuntu 20l04

when running again this examle 

 cargo run --example cxx-ros2-dataflow --features ros2-examples

I got 

/home/demo/Public/dora/examples/c++-ros2-dataflow/node-rust-api/main.cc:23:28: error: no member named 'create_topic_geometry_msgs_Twist' in 'Ros2Node'
    auto vel_topic = node->create_topic_geometry_msgs_Twist("/turtle1", "cmd_vel", qos);
                     ~~~~  ^
/home/demo/Public/dora/examples/c++-ros2-dataflow/node-rust-api/main.cc:24:32: error: no member named 'create_publisher' in 'Ros2Node'
    auto vel_publisher = node->create_publisher(vel_topic, qos);
                         ~~~~  ^
/home/demo/Public/dora/examples/c++-ros2-dataflow/node-rust-api/main.cc:25:29: error: no member named 'create_topic_turtlesim_Pose' in 'Ros2Node'
    auto pose_topic = node->create_topic_turtlesim_Pose("/turtle1", "pose", qos);
                      ~~~~  ^
/home/demo/Public/dora/examples/c++-ros2-dataflow/node-rust-api/main.cc:26:36: error: no member named 'create_subscription' in 'Ros2Node'
    auto pose_subscription = node->create_subscription(pose_topic, qos, merged_events);
                             ~~~~  ^
/home/demo/Public/dora/examples/c++-ros2-dataflow/node-rust-api/main.cc:55:17: error: use of undeclared identifier 'geometry_msgs'
                geometry_msgs::Twist twist = {
                ^
5 errors generated.

bobd988, 2024-03-29:
DUP as #443

phil-opp, 2024-03-30:
Issue #443 is about a SIGABRT error when running the `ros2-service-server` branch, which is not merged yet. So I don't think that this issue is related to it.

>  no member named 'create_topic_geometry_msgs_Twist' in 'Ros2Node'

This sounds like you maybe forgot to [source your ROS2 installation](https://docs.ros.org/en/humble/Tutorials/Beginner-CLI-Tools/Configuring-ROS2-Environment.html#source-the-setup-files) before running the example?

bobd988, 2024-03-30:
it is environment  related. I will try another PC

phil-opp, 2024-03-30:
Is the `ros2` command available in the terminal where you run the example?

haixuanTao, 2024-08-31:
Closing this as it seems solved.

closed 2024-08-31 by haixuanTao