dora-rs/dora #459: Tracking issue for distributed dora dataflow

haixuanTao, 2024-04-07:
**Describe the bug**
Dora remote connection has been long awaited. We however need to fix a couple of things to make it work:

**Tracking issue**
- [ ] Make hard-coded localhost addr configurable in the daemon and the coordinator
- [ ] Make `working_dir` configurable dependent on the machine on multi daemon situation.

haixuanTao, 2024-04-08:
Yeah sorry could have been more precise:
- https://github.com/dora-rs/dora/blob/9508fbb399ed6722a13497cfddc7d0441d64cdd0/binaries/daemon/src/inter_daemon.rs#L66-L88
- https://github.com/dora-rs/dora/blob/9508fbb399ed6722a13497cfddc7d0441d64cdd0/binaries/coordinator/src/listener.rs#L9-L20

phil-opp, 2024-04-08:
Ah yes, we should use `0.0.0.0` in the coordinator to listen for incoming connections on all interfaces. For the `daemon`, the `127.0.0.1` should be fine since only local nodes should initiate connections to the daemon.

haixuanTao, 2024-04-08:
I have just looked rapidly but isn't this: - https://github.com/dora-rs/dora/blob/9508fbb399ed6722a13497cfddc7d0441d64cdd0/binaries/daemon/src/inter_daemon.rs#L66-L88 
for inter_daemon connection?

Michael-J-Ward, 2024-04-11:
>  Make working_dir configurable dependent on the machine on multi daemon situation.

Should `dora` attempt to create the `working_dir` if it doesn't exist or fail out with an error?

> Ah yes, then we should listen on 0.0.0.0 in the deamon as well.

Should `0.0.0.0` be the default, or should it not be configurable? (I don't have experience networking robots, but I'd assume connecting them through a VPN would be common - `cargo run -- --bind <vpn-interface-ip>:<port>`)

haixuanTao, 2024-04-11:
> > Make working_dir configurable dependent on the machine on multi daemon situation.
> 
> Should `dora` attempt to create the `working_dir` if it doesn't exist or fail out with an error?

So there is couple of things that we consider: https://github.com/dora-rs/dora/pull/256#issuecomment-1514362986

I think that we could maybe as a first iteration expect the working_dir to exist with the right files in it, and error out on missing files. We can in a follow up PR handle creating missing folder.

For more, see: https://github.com/dora-rs/dora/pull/256#issuecomment-1514362986

> 
> > Ah yes, then we should listen on 0.0.0.0 in the deamon as well.
> 
> Should `0.0.0.0` be the default, or should it not be configurable? (I don't have experience networking robots, but I'd assume connecting them through a VPN would be common - `cargo run -- --bind <vpn-interface-ip>:<port>`)

It does seem like being configurable would be a plus. I think that by revisiting the port argument it should be fairly straightforward to make addr configurable.

Michael-J-Ward, 2024-04-11:
> As far as I know, 0.0.0.0 acts as a kind of wildcard when listening. So it should listen for incoming connections on all available interfaces.

Correct- binding to `0.0.0.0` will listen all interfaces available to the process.

Binding to `<vpn-net-ip>` commonly *restricts* connections to only allow other machines on the VPN.

phil-opp, 2024-04-11:
Ah, that makes sense, thanks for clarifying! I agree that making this configurable is a good idea then.

Michael-J-Ward, 2024-04-17:
Many of the `cli` commands end up using the `connect_to_coordinator` function.

Should all of these be made configurable as well?

https://github.com/dora-rs/dora/blob/1c2dc46ac272ce607200f089e3c43cdf76daf2a8/binaries/cli/src/main.rs#L435-L437

Michael-J-Ward, 2024-04-17:
Reupping the above after taking a closer look at the `daemon` CLI command, which explicitly *ignores* the coordination addr when startingn a dataflow and advises

> Please use the `start` command for remote coordinator

But `start`   uses the default `connect_to_coordinator` from above, not using any passed arguments.

https://github.com/dora-rs/dora/blob/1c2dc46ac272ce607200f089e3c43cdf76daf2a8/binaries/cli/src/main.rs#L291-L313

phil-opp, 2024-04-18:
Yes, thanks a lot for all your work on this!

The correct solution for the CLI would be to add a `coordinator_addr` argument to the `connect_to_coordinator` function. We would also need some way to set this argument when running a `dora` CLI command. I think we can start with a new command-line argument for this and look for more convenient ways in the future. What do you think?

bobd988, 2024-05-16:
I received feedback from two serious customers that Zenoh support across networks is a key differentiator feature.

haixuanTao, 2024-05-16:
Then we can prioritise it after https://github.com/dora-rs/dora/pull/478 and https://github.com/dora-rs/dora/pull/497 and https://github.com/dora-rs/dora/pull/495

phil-opp, 2024-06-05:
I opened an issue with the next required steps for distributed dataflows: https://github.com/dora-rs/dora/issues/535. Discussion and pull requests are welcome!