dora-rs/dora #481: self-defined messages not able to included in header file

bobd988, 2024-04-20:
For self-defined ROS2 msg , after adding to AMENT_PREFIX_PATH , and then run cargo build the msg was not inlcuded in the bindings header file. 

The expected behavior is to having the message to be included in  bindings header file if ROS2 msg folder was added AMENT_PREFIX_PATH . By the way the msg folder by default has no idl file. Is idl  file needed in order to make this work?

I was running this command after adding msg to AMENT_PREFIX_PATH 

export RMW_IMPLEMENTATION=rmw_fastrtps_cpp
cargo run --example cxx-ros2-dataflow --features ros2-examples

phil-opp, 2024-04-21:
Our current message generator looks at the files in `share/ament_index/resource_index/rosidl_interfaces` to find all available messages. Then it looks up the listed messages in `share/<namespace>/msg/<name>.msg`. I think we could skip the lookup in `rosidl_interfaces` and look for the message files directly.

I'm not familiar with self-defined ROS2 messages. Could you give some details on the directory structure so that I can update the message generator to work for your case?

bobd988, 2024-04-24:
Ubuntu 22.04  OS switch to fix-ament-prefix-path branch and 

`cargo run --example cxx-ros2-dataflow --features ros2-examples`
 
I got compile error 

  Compiling dora-node-api v0.3.3 (/home/demo/Public/github_dora/dora/apis/rust/node)
error: failed to run custom build command for `dora-ros2-bridge v0.1.0 (/home/demo/Public/github_dora/dora/libraries/extensions/ros2-bridge)`

Caused by:
  process didn't exit successfully: `/home/demo/Public/github_dora/dora/target/debug/build/dora-ros2-bridge-3496c43697944aee/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
  cargo:rerun-if-changed=/opt/ros/humble
  cargo:rerun-if-changed=
  cargo:rerun-if-changed=/home/demo/msgs
  cargo:rerun-if-changed=/home/demo/msgs/autoware_msgs/autoware_common_msgs/msg/

  --- stderr
  GlobError { path: "/boot/efi", error: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } }
  GlobError { path: "/etc/cups/ssl", error: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } }
  GlobError { path: "/etc/polkit-1/localauthority", error: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } }
  GlobError { path: "/etc/ssl/private", error: Os { code: 13, kind: PermissionDenied, message: "Permission denied" } }
  thread 'main' panicked at /home/demo/Public/github_dora/dora/libraries/extensions/ros2-bridge/msg-gen/src/lib.rs:23:40:
  called `Result::unwrap()` on an `Err` value: No such file or directory (os error 2)
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
Error: failed to compile dora-node-api-cxx

Location:
    examples/c++-ros2-dataflow/run.rs:74:9

haixuanTao, 2024-04-26:
Indeed! Thanks for trying. Just pushed a fix on the branch.

PR #489 (2024-05-01): Simplify parsing of `AMENT_PREFIX_PATH`
This PR simplify how ROS2 messages are parsed using dora.

It will parse all messages included in the `AMENT_PREFIX_PATH` folder that has a path in the following format: `<package>/msg/<message>.msg`.

Similarly for services it will be parsed as: `<package>/srv/<service>.srv`.

It does not requires `.idl` anymore nor does it needs to be in the `ament_index`.

Fixes: #481
closed 2024-05-01 by haixuanTao