dora-rs/dora #483: ROS2 messaages not work between python and C++ dora node

bobd988, 2024-04-20:
When using DORA C++ with the ROS2 messages to pub/sub 

[image]

Using another DORA node with python to subscribe to it but message not able to receive.  If you can try it works pls provide an example.

haixuanTao, 2024-04-21:
The ros2 bridge in dora is provided mainly thanks to https://github.com/jhelovuo/ros2-client .

Also, we're able to communicate with ros2, there should still be expected errors when using the bridges as both project are fairly young and experimental. 

Using the bridge between two dora nodes is not in the range of things we can actively support, because:
- we don't own ros2-client and most of the logic happens there.
- there is a lot of moving parts between dora <-> DDS <-> ros2 which means a lot of errors can happen.
- we still need to actively develop dora and it's missing features. We barely support all dora features in C++ so actively pushing on the C++ ros2 bridge before dora seems precipitated.

I would recommend switching both nodes to dora, using arrow.

phil-opp, 2024-04-21:
You could try to test the two nodes separately against ROS2 nodes:

1. Verify that ROS2 messages sent by the Dora C++ node are seen by ROS2.
2. Verify that the Python Dora node receives messages sent by ROS2 nodes.

If both of the above work, I would expect that a Dora C++ publish to ROS2 should reach a Dora Python subscription to that same ROS2 topic.