dora-rs/dora #502: Error when a node receives two data streams

RuPingCen, 2024-05-18:
We try to use node C to receive data streams sent by two different nodes A and B. **Sometimes** the receiving node C will have the error shown in the figure below,

**[c node] received unexpected event: 2**

[
[image]
](url)

This results in the functions on lines 126 and 130 of the code never being called, as shown in the figure below

 
[image]

Here is the dataflow.yaml

```
- id: controltest
    custom:
      source: controldata
      inputs:
        tick: dora/timer/millis/100
      outputs:
        - road_lane
      
  - id: road_lane
    custom:
      source: ../../map/road_line_publisher/build/road_lane_publisher_node
      inputs:
              DoraGnssPose: gnss_poser_sub/DoraGnssPose
              road_lane: controltest/road_lane
      outputs:
            - cur_pose_all
            
  - id: planning
    custom:
      source: ../../planning/routing_planning/build/routing_planning_node
      inputs:
              cur_pose_all: road_lane/cur_pose_all
              road_lane: controltest/road_lane
      outputs:
            - raw_path
```

phil-opp, 2024-05-20:
Some nodes seem to exit with a `-1` return value. What is the cause for that?

> [c node] received unexpected event: 2

Event type `2` is an `InputClosed` event. Dora sends this event when one of a node's inputs is closed. Inputs are closed when the source node exits.

RuPingCen, 2024-05-23:
Thanks for your reply. I wrote a c_node test example.

https://github.com/dora-rs/autoware.universe/commit/e10fd19a7140492f36bc05a395eebec18c006338

Here, nodes A and B publish data at 10Hz and 100Hz respectively. Node C receives data from nodes A and B, and no such error occurs. This does not seem to be a dora kernel error.

I will continue to check my code according to your suggestion.

phil-opp, 2024-05-23:
Could you share the full output from your `out` file as well please? I tried to run your example code but I don't see any errors apart from some "stream did not contain valid UTF-8" errors.

Hennzau, 2025-03-18:
I'm closing it as it seems solved

closed 2025-03-18 by Hennzau