dora-rs/dora #504: Make dora node API minor versions compatible with one another.

haixuanTao, 2024-05-21:
**Describe the bug**
Make dora API minor versions compatible with one another.

https://github.com/dora-rs/dora/blob/dae290f83ced1f60c8f89e1ad1f5e0b0cfb37149/binaries/daemon/src/node_communication/mod.rs#L193-L198

Currently we force users to bump all existing dora node to use the latest API version. This can be challenging especially when using Rust where we need to modify a lot of Cargo.toml.

Could be good to have non breaking minor versions that are compatible with one another.

phil-opp, 2024-05-22:
We're still at 0.3.4, so we treat minor version bumps as breaking changes (e.g. 0.3.4 -> 0.4.0). So I assume that you mean patch version bumps (e.g. 0.3.4 -> 0.3.5)?

I'm absolutely in favor of making the message formats between nodes, daemons, and the coordinator more stable and backwards compatible. However, changes to the message format become more difficult then because all non-compatible changes need to wait until 0.4.0 then.

Maybe we could move all the messaging types into a common sub-crate that we release and version independently of the main crates. This way, we could compare the versions of this sub-crate to check if messages are compatible, so we would be independent of the main version numbers.

haixuanTao, 2024-05-22:
Yes patch version.

I think it's fine to have breaking patch version changes as we're still experimental, but the default should be to allow multiple versions for each daemon rather than defaulting to make an upgrade.

phil-opp, 2024-05-22:
Then we need some kind of separate versioning as I proposed above. We need some way to figure out whether nodes/daemons/coordinator are compatible before trying to parse the messages.

haixuanTao, 2024-05-22:
Ok, let's not reinvent the wheel, let's do:
- path version are compatible with one another
- minor version introduce breaking changes

phil-opp, 2024-05-22:
I think that's a good policy, but I'm not sure if we want to implement this just yet. We might still require some breaking changes to the message format in the near future, e.g. for #459. Right now we could implement this in a new patch version, but once we change our compatibility guarantee we would need to postpone such changes until we release the next minor version.

closed 2024-08-26 by haixuanTao