dora-rs/dora #511: Make dora cli connect to remote coordinator

haixuanTao, 2024-05-24:
- [ ] Make CLI coordinator connection configurable.

Currently the CLI can only connect to the local preconfigured coordinator. We should make it configurable such as:

Ex:
```bash
dora start dataflow.yaml --coordinator-addr IP:PORT
``` 

https://github.com/dora-rs/dora/blob/ea47a556a77f6c64f29ab07939d747edc8573671/binaries/cli/src/main.rs#L469-L472

Gege-Wang, 2024-05-25:
There are three questions:
1. Does dora have only one coordinator? 
2. Does dora start passed --coordinator-addr first or dora up pass first? 
dora start and dora destroy both use connect_to_coordinator(). 
3. If we specify only one coordinator, do we need to use static variables to store the address parameters?

phil-opp, 2024-06-05:
This has been fixed by:

- https://github.com/dora-rs/dora/pull/513
- https://github.com/dora-rs/dora/pull/520

closed 2024-06-05 by phil-opp