dora-rs/dora #516: Error when using c_node

starlitxiling, 2024-05-28:
**Describe the bug**
I tried to send and receive messages using the C API in dora, but it gave some errors. I found that the init_dora_context_from_env or dora_next_event function had a problem, and then I found that DORA_NODE_CONFIG was not set. I don't know how dora sets DORA_NODE_CONFIG according to the yml file during operation, but this problem did not occur in the 0.3.2 version of dora.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or Video**
[image]
[image]

**Environments**
 - System info: Ubuntu 20.04
 - Dora version: 0.3.4

haixuanTao, 2024-05-28:
`DORA_NODE_CONFIG` is an environment variable that is set when dora spawn a node in `dora start` you should not set it yourself.

If this issue happens, there is probably something wrong with how you start the dataflow or the environment variables that are being passed to your process. 

It seems to be fine in our CI.

Would you mind trying what the difference is between the CI and your node, in this very specific context?

starlitxiling, 2024-05-29:
Yes, I tried to run some C examples according to CI, and found that sometimes it does not have the above error, but sometimes it will have the error that `DORA_NODE_CONFIG` is not set. I don't know why, my system and dora version have not changed between them.

phil-opp, 2024-05-29:
Could you share a link to your code? It's a bit unclear to me how a segfault could occur inside the `poll_next` method of some channel library.

starlitxiling, 2024-05-29:
I just tried this https://github.com/dora-rs/autoware.universe/tree/feature/autoware_dora/tools/C_node_test and c++-dataflow in dora's example. Regarding segfault, I felt during debugging that `init_dora_context_from_env` did not initialize a pointer correctly, and then an error occurred in `dora_next_event`. Now my dora version has been changed to 0.3.3, and this problem does not occur.

phil-opp, 2024-05-29:
Thanks for the update! So the problem appears to be fixed after updating to `dora v0.3.3`?

starlitxiling, 2024-05-29:
When my dora version is 0.3.4, it sometimes has this problem. The repo branch of the c library I use is the main branch. When I switch to tags/v0.3.3, the problem does not occur.

bobd988, 2024-05-30:
I met same issue after I upgraded to 0.3.4 last night

phil-opp, 2024-06-06:
I tried to reproduce this a few times with the `C_Python_test` example as suggested in https://github.com/dora-rs/dora/issues/515#issuecomment-2140087939, but this error never happened for me.

starlitxiling, 2024-06-10:
Yes, it only appears sometimes, can you tell me your system information and dora version, and which branch of the dora repo is it (I don't know if this has anything to do with dora-node-api.h or something)

starlitxiling, 2025-03-19:
I think this problem is caused by `conda`. `dora up` and `dora start` are not in the same `conda` environment. I used `conda`carefully and the above problem did not occur. I think this issue has been resolved.

haixuanTao, 2025-03-19:
That would make sense. Thanks for the update :)

closed 2025-03-19 by haixuanTao