dora-rs/dora #530: Hot reload does not work with complex configuration vim

hbuxiaofei, 2024-06-04:
Env:
- Ubuntu 24.04 LTS
- dora-cli 0.3.4

When editing a source file using the vim editor, the following events will occur:
```
>>> recv an event: Ok(Event { kind: Modify(Name(From)), paths: ["/home/rayylee/github/code-test/rust/dora-test/python-operator-dataflow/plot.py"], attr:tracker: None, attr:flag: None, attr:info: None, attr:source: None })
>>> recv an event: Ok(Event { kind: Modify(Metadata(Any)), paths: ["/home/rayylee/github/code-test/rust/dora-test/python-operator-dataflow/plot.py"], attr:tracker: None, attr:flag: None, attr:info: None, attr:source: None })
>>> recv an event: Ok(Event { kind: Remove(File), paths: ["/home/rayylee/github/code-test/rust/dora-test/python-operator-dataflow/plot.py"], attr:tracker: None, attr:flag: None, attr:info: None, attr:source: None })
```
The `ModifyKind::Data` event will not occur.

phil-opp, 2024-06-04:
Thanks for reporting!

Apparently vim is replacing the file with a new one when saving it, instead of modifying it. See https://github.com/notify-rs/notify/issues/247 and https://github.com/notify-rs/notify/issues/113. So it looks like we have to watch the directory instead.

phil-opp, 2024-06-04:
I opened a PR to hopefully fix this issue: https://github.com/dora-rs/dora/pull/532. Could you please try whether it works for you? Thanks!