dora-rs/dora #540: Thread-safety of C node API

phil-opp, 2024-06-07:
We should clearly document, which parts of the C node API are thread-safe and which not. Also, we should try to make more functions thread-safe, if possible.

Example: The `dora_next_event` function casts the given context pointer to a `&mut DoraContext` in Rust, which must be an exclusive reference. So calling the function concurrently with the same context pointer violates this requirement and leads to undefined behavior. However, the underlying `flume` channel supports receiving elements through shared references, so we might be able to make `dora_next_event` thread-safe.

SanjayUG, 2025-03-19:
Hi there,

I would like to work on it,

@phil-opp please assign me for the contribution.

Thank You.

Hennzau, 2025-03-19:
> Hi there,
> 
> I would like to work on it,
> 
> [@phil-opp](https://github.com/phil-opp) please assign me for the contribution.
> 
> Thank You.

you're assigned to it!