dora-rs/dora #541: Some issues we are currently facing with using dora-rs.

chengzi0103, 2024-06-08:
1. Dora does not seem to support multi-virtual environment control in conda

When running dora start in WSL, everything runs in the base environment. If you are using conda environments, you must develop in the base environment as it does not support multiple environments.

2. Unable to debug when running dora-start directly

If you run dora start directly, you cannot debug the Python code (as it starts in cmd mode).

3. Does not seem to support manual input in the workflow？
If there is an input command in the Python code, it will cause an error during execution.

4. Can one dataflow node call/embed another dataflow?

5. Does Dataflow support conditional judgment and branching?

6. Does Input support variable input？ Can Input get content input from the terminal window?

haixuanTao, 2024-06-10:
> -   Dora does not seem to support multi-virtual environment control in conda
Yes, it's a bit of a stretch because conda is a bit of a mess but you can use absolute python path:
```yaml
  - id: eval
    custom:
      source: /home/antoine/Pollen/lerobot/lerobot/scripts/eval.py
      args: -p cadene/2024_06_06_act_reachy2_mobile_base_3_060000 eval.n_episodes=1 eval.batch_size=1 env.episode_length=20000 policy.n_action_steps=100
```

Use conda run:
```
  - id: custom-node_1
    custom:
      source: conda
      args: run -n my-python-env node.py
```
Note that conda run can sometimes be slow for unknown reasons. 

But I agree that it is not super explicit.

>     Unable to debug when running dora-start directly
 
I'm working on making possible to #517  run nodes dynamically by just calling:
```
python your_node.py
```

And also make logs more explicit with #542 

> Does not seem to support manual input in the workflow？

This would be fixed by #517 

> Can one dataflow node call/embed another dataflow?

It doesn't and might not in the short term

> Does Dataflow support conditional judgment and branching?

We don't and probably won't 

> Does Input support variable input？ Can Input get content input from the terminal window?

By variable, do you mean different types? If so, yes, we use arrow types and you can match the types as input.

If you mean variable input source, we don't do this for the moment.

> Can Input get content input from the terminal window?

So we're going to work on this after #517  probably but not in the short term.

Sorry, if we're not were you would expect us to be, but be sure that we're working really hard to be there. But, thanks for your interest!

haixuanTao, 2024-07-25:
Can I close this or do you guys still need help on specific point that `dynamic_node` does not fix?

chengzi0103, 2024-08-02:
> dynamic_node

> Yes, it's a bit of a stretch because conda is a bit of a mess but you can use absolute python path:
>
> ```yaml
>   - id: eval
>     custom:
>       source: [...]

Sure, here is the translation: "Of course it can be turned off. Thank you for your answer."

chengzi0103, 2024-08-02:
> Can I close this or do you guys still need help on specific point that `dynamic_node` does not fix?
Of course it can be turned off. Thank you for your answer.

closed 2024-08-11 by Hennzau