dora-rs/dora #564: Musl C lib support

bobd988, 2024-06-24:
This may not be Dora specific, but there is a request to support OpenHarmony 4.x (OHOS). OHOS is not using glibc but musl C.  Is there any solution that Dora can support operating system that uses musl C.

phil-opp, 2024-06-24:
Have you tried compiling dora for a musl target, e.g. for `x86_64-unknown-linux-musl`? I just tried the following and it worked without error:

```
> rustup target install x86_64-unknown-linux-musl
> cargo build -p dora-cli --target x86_64-unknown-linux-musl
```

XxChang, 2024-06-29:
I compile it with target "x86_64-unknown-linux-ohos", then I get an error
```
error[E0412]: cannot find type `IovLen` in this scope
   --> /home/xuchang/.cargo/registry/src/mirrors.ustc.edu.cn-4affec411d11e50f/socket2-0.4.10/src/sys/unix.rs:837:63
    |
837 |     msg.msg_iovlen = min(bufs.len(), IovLen::MAX as usize) as IovLen;
    |                                                               ^^^^^^ not found in this scope

error[E0412]: cannot find type `IovLen` in this scope
   --> /home/xuchang/.cargo/registry/src/mirrors.ustc.edu.cn-4affec411d11e50f/socket2-0.4.10/src/sys/unix.rs:897:63
    |
897 |     msg.msg_iovlen = min(bufs.len(), IovLen::MAX as usize) as IovLen;
    |                                                               ^^^^^^ not found in this scope

error[E0433]: failed to resolve: use of undeclared type `IovLen`
   --> /home/xuchang/.cargo/registry/src/mirrors.ustc.edu.cn-4affec411d11e50f/socket2-0.4.10/src/sys/unix.rs:837:38
    |
837 |     msg.msg_iovlen = min(bufs.len(), IovLen::MAX as usize) as IovLen;
    |                                      ^^^^^^ use of undeclared type `IovLen`

error[E0433]: failed to resolve: use of undeclared type `IovLen`
   --> /home/xuchang/.cargo/registry/src/mirrors.ustc.edu.cn-4affec411d11e50f/socket2-0.4.10/src/sys/unix.rs:897:38
    |
897 |     msg.msg_iovlen = min(bufs.len(), IovLen::MAX as usize) as IovLen;
    |                                      ^^^^^^ use of undeclared type `IovLen`

```

It seems like that `socket2-0.4.10` had not supported ohos yet. Too many crates are depend on it.

In addition, to support ohos, we should modify the `rust-toolchain.toml`. `rust-1.76` does not support `x86_64-unknown-linux-ohos`.

phil-opp, 2024-07-01:
> It seems like that `socket2-0.4.10` had not supported ohos yet (these should be supported in `socket2-0.5.*`). Too many crates are depend on it.
> 
> Please check [rust-lang/socket2#451](https://github.com/rust-lang/socket2/issues/451)

Ok, looks like we need to update some dependencies then.

> In addition, to support ohos, we should modify the `rust-toolchain.toml`. `rust-1.76` does not support `x86_64-unknown-linux-ohos`.

Is the version too new or too old? The target was addedd in https://github.com/rust-lang/rust/pull/113061 , so it should be available in 1.76. If it's too old, you can easily compile the latest stable version using `cargo +stable build` or `cargo +stable install`. We don't want to change the `rust-toolchain.toml` currently because we don't want to accidentally use features that force users to use the very latest Rust version.

XxChang, 2024-07-01:
> Is the version too new or too old? The target was addedd in [rust-lang/rust#113061](https://github.com/rust-lang/rust/pull/113061) , so it should be available in 1.76. If it's too old, you can easily compile the latest stable version using `cargo +stable build` or `cargo +stable install`. We don't want to change the `rust-toolchain.toml` currently because we don't want to accidentally use features that force users to use the very latest Rust version.

I just type `rustup target list | grep ohos`, nothing works with `rust 1.76`.

phil-opp, 2024-07-01:
Ah, looks like the stdlib is only available since https://github.com/rust-lang/compiler-team/issues/719, which was released with rust 1.78. The `cargo +stable` commands I mentioned above should work though.

Gege-Wang, 2024-07-04:
for rk3568 is armv7, we should use musl libc.
```bash
docker pull messense/rust-musl-cross:arm-musleabi
docker run --rm -it -v "$(pwd)":/home/rust/src messense/rust-musl-cross:arm-musleabihf bash
rustup target add arm-unknown-linux-musleabi
cargo build -p dora-cli --release --target arm-unknown-linux-musleabi
```
so we can move ./dora/target/arm-unknown-linux-musleabi/release/dora to rk3568.
another note: we should use dynamic link with C node
```bash
/path/to/dora/.cargo/config.toml
[target."armv7-unknown-linux-musleabi"]
rustflags = "-C target-feature=-crt-static"
```
and move libgcc_s.so.1 to rk3568:/lib
```
# in above docker container
the libgcc_s.so.1 path is  /usr/local/arm-linux-musleabi/lib/libgcc_s.so.1 
```

closed 2024-07-04 by Gege-Wang

haixuanTao, 2024-07-04:
I can add arm-musl in our next github release