dora-rs/dora #565: How to insert manual input during the execution of Dora?

chengzi0103, 2024-06-25:
**Describe the bug**
I want to determine during the execution of a Dora workflow whether some content needs manual intervention based on certain conditions. Following the example from `https://github.com/dora-rs/dora/blob/main/examples/python-operator-dataflow/keyboard_op.py`, I built a workflow myself. However, during the execution of the workflow, manual input was not integrated, and I am not sure where the issue lies. Additionally, I noticed that in your example, after using `keyboard_op.py`, manual input is possible. I want to know how I can solve this problem.

**Expected behavior**

**Screenshots or Video**
[image]

**Environments (please complete the following information):**
 - System info: Wsl-Ubuntu 22.04 
 - Dora version: 0.3.4
 - Python version: 3.11.9

**Additional context**
I have uploaded my workflow files in a compressed folder. Could you please help check if there are any issues?
[keyboard.zip](https://github.com/user-attachments/files/15963578/keyboard.zip)

DuinoDu, 2024-06-25:
I tried your keyboard.zip on mac and keyboard input can be captured by dora.

https://github.com/dora-rs/dora/assets/6895453/32fc24b6-4d6e-4b3d-827a-01cc6b9e494d

Is that what you expect?

I also tried on wsl2-ubuntu, but found error when install pynput. [pynput seems not working well on wsl2.](https://github.com/moses-palmer/pynput/issues/286)

haixuanTao, 2024-06-26:
We're going to merge #559  and later release a release candidate for 0.3.5 that you will be able to try with dynamic nodes.

chengzi0103, 2024-07-04:
> I tried your keyboard.zip on mac and keyboard input can be captured by dora.
> 
>  2024-06-25.15.32.37.mov 
> Is that what you expect?
> 
> I also tried on wsl2-ubuntu, but found error when install pynput. [pynput seems not working well on wsl2.](https://github.com/moses-palmer/pynput/issues/286)

Thank you for your reply, I apologize for the late response.
In our WSL/VM environment, we are unable to control hardware using Pynput.
I suspect this is due to security considerations in these environments, as they typically do not allow direct access to the underlying hardware. This is what leads to our inability to use them in the virtual environment.
Below is the error message we received:

~~~
Traceback (most recent call last):
  File "/tmp/cc/experiment_project/experiment_project/moxin/keyboard/keyboard_op.py", line 1, in <module>
    from pynput import keyboard
  File "/root/miniconda3/lib/python3.12/site-packages/pynput/__init__.py", line 40, in <module>
    from . import keyboard
  File "/root/miniconda3/lib/python3.12/site-packages/pynput/keyboard/__init__.py", line 31, in <module>
    backend = backend(__name__)
              ^^^^^^^^^^^^^^^^^
  File "/root/miniconda3/lib/python3.12/site-packages/pynput/_util/__init__.py", line 77, in backend
    raise ImportError('this platform is not supported: {}'.format(
ImportError: this platform is not supported: ('failed to acquire X connection: Bad display name ""', DisplayNameError(''))

Try one of the following resolutions:

 * Please make sure that you have an X server running, and that the DISPLAY environment variable is set correctly
~~~

haixuanTao, 2024-07-04:
I understand could you try dynamic node: https://github.com/dora-rs/dora/releases/tag/v0.3.5 ?

chengzi0103, 2024-07-04:
> I understand could you try dynamic node: https://github.com/dora-rs/dora/releases/tag/v0.3.5 ?

Okay, I have tried it. When I used dora-cli with version 0.3.5-rc0, and I updated my Python-dora through the command pip install dora-rs==0.3.5 --force, it resulted in an error. I don't know what caused it. Currently, I am still testing in a VM (Virtual Machine) environment
~~~

[ERROR]
Dataflow could not be validated.
~~~
[image]

chengzi0103, 2024-07-04:
> > I understand could you try dynamic node: https://github.com/dora-rs/dora/releases/tag/v0.3.5 ?
> 
> Okay, I have tried it. When I used dora-cli with version 0.3.5-rc0, and I updated my Python-dora through the command pip install dora-rs==0.3.5 --force, it resulted in an error. I don't know what caused it. Currently, I am still testing in a VM (Virtual Machine) environment
> 
> ```
> 
> 
> [ERROR]
> Dataflow could not be validated.
> ```
> 
> ![dora-rc-error](https://private-user-images.githubusercontent.com/23193969/345730882-6accd037-69c1-44eb-93f6-c0827c42d9aa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAwNzcyMDEsIm5iZiI6MTcyMDA3NjkwMSwicGF0aCI6Ii8yMzE5Mzk2OS8zNDU3MzA4ODItNmFjY2QwMzctNjljMS00NGViLTkzZjYtYzA4MjdjNDJkOWFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzA0VDA3MDgyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk4YjViMmFiYjcwYzIzMzUzMTNkZTExNmUwMDEyM2U4NjQ3MDc5ODI5NGFlNTMxNmEwMDNkM2UwNGU0NmQ5NzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.AfL6yu5km5IcUAYdYYKCzM0YLoq1Y1a5V6WJ77PYcKc)

When I reinstalled dora-cli using cargo install dora-cli --locked, the same issue occurred

haixuanTao, 2024-07-04:
Could you run:
```bash
cargo install dora-cli==0.3.5 --locked
pip install dora-rs==0.3.5
```
?

chengzi0103, 2024-07-04:
> ```shell
> pip install dora-rs==0.3.5
> ```

error: could not find `dora-cli==0.3.5` in registry `crates-io` with version `*`

haixuanTao, 2024-07-04:
You can also use:
```
export DORA_VERSION=v0.3.5 # Check for the latest release
export ARCHITECTURE=$(uname -m)
wget https://github.com/dora-rs/dora/releases/download/${DORA_VERSION}/dora-${DORA_VERSION}-${ARCHITECTURE}-Linux.zip
unzip dora-${DORA_VERSION}-${ARCHITECTURE}-Linux.zip
pip install dora-rs==${DORA_VERSION}
PATH=$PATH:$(pwd)
dora --help

```

chengzi0103, 2024-07-04:
> You can also use:您还可以使用：用途：
> 
> ```
> export DORA_VERSION=v0.3.5 # Check for the latest release
> export ARCHITECTURE=$(uname -m)
> wget https://github.com/dora-rs/dora/releases/download/${DORA_VERSION}/dora-${DORA_VERSION}-${ARCHITECTURE}-Linux.zip
> unzip dora-${DORA_VERSION}-${ARCHITECTURE}-Linux.zip
> pip install dora-rs==${DORA_VERSION}
> PATH=$PATH:$(pwd)
> dora --help
> ```

This is the version information below.
[image]

chengzi0103, 2024-07-04:
> Sorry you need to do:对不起，您需要做：
> 
> ```shell
> dora destroy
> dora up
> ```
> 
> After a version update版本更新后

Okay, thank you. The process has started normally

chengzi0103, 2024-07-04:
> I understand could you try dynamic node: https://github.com/dora-rs/dora/releases/tag/v0.3.5 ?我知道你能试试动态节点：https://github.com/dora-rs/dora/releases/tag/v0.3.5吗？

When I use the example from [https://github.com/dora-rs/dora-rs.github.io/blob/v0.3.5/docs/guides/Development/dynamic-node.md], I encountered several issues:

I cannot terminate the dataflow using dora stop.
After using dora destroy && dora up to restart the dataflow, and then using:
from dora import Node
node = Node("node_0")
event = node.next()
print(event)
I still cannot retrieve the messages corresponding to the node_id
~~~
>>> from dora import Node
>>> node = Node("node_0") 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Could not setup node from node id. Make sure to have a running dataflow with this dynamic node

Caused by:
    failed to get node config from daemon: no node with ID `{node_id}`

Location:
    apis/rust/node/src/node/mod.rs:101:17

~~~

This is the dataflow I am currently running
~~~
dora list 
UUID                                  Name     Status
01907ce7-4513-7003-af5a-ef13a79d37db  dynamic  Running

~~~

haixuanTao, 2024-07-04:
So our CI is passing on this:

https://github.com/dora-rs/dora/blob/2f401ff8464aba4e074206d1a122ce4ee4d08e4f/.github/workflows/ci.yml#L322-L323

Could you check if you can run this from this repo:

```bash
pip install "numpy<2.0.0" opencv-python
dora start ../examples/python-dataflow/dataflow_dynamic.yml --name ci-python-dynamic --detach
python ../examples/python-dataflow/plot_dynamic.py
```

chengzi0103, 2024-07-05:
> So our CI is passing on this:所以我们的线人会告诉你
> 
> https://github.com/dora-rs/dora/blob/2f401ff8464aba4e074206d1a122ce4ee4d08e4f/.github/workflows/ci.yml#L322-L323
> 
> Could you check if you can run this from this repo:您能否检查是否可以从该存储库运行此操作：
> 
> ```shell
> pip install "numpy<2.0.0" opencv-python
> dora start examples/python-dataflow/dataflow_dynamic.yml --name ci-python-dynamic --detach
> python examples/python-dataflow/plot_dynamic.py
> ```

Thank you for your reply. After I restarted the process, I successfully ran your example. Thanks. But now I have another issue: How do I stop this dataflow? When I use dora stop 'dataflow_id', it gets stuck

haixuanTao, 2024-07-05:
I wonder if this is linked to #575 . 

You are on WSL right?

chengzi0103, 2024-07-05:
> I wonder if this is linked to #575 .
> 
> You are on WSL right?

Yes, I have tried both in WSL and VM, and this problem has occurred in both cases. System version: Ubuntu 22.04    
I think it is consistent with the issue in  https://github.com/dora-rs/dora/issues/575

haixuanTao, 2024-07-05:
Ok, interesting.

For `examples/python-dataflow/dataflow_dynamic.yml` after 10 second, the dataflow should stop on its own.

Could you try to set the same logic in your usecase. I am going to fix #575 on a follow up PR.

chengzi0103, 2024-07-10:
> Ok, interesting. 好吧，有趣。
> 
> For `examples/python-dataflow/dataflow_dynamic.yml` after 10 second, the dataflow should stop on its own.对于 10 秒后的 `examples/python-dataflow/dataflow_dynamic.yml` ，数据流应自行停止。
> 
> Because of: 因为：
> 
> https://github.com/dora-rs/dora/blob/2f401ff8464aba4e074206d1a122ce4ee4d08e4f/examples/python-dataflow/webcam.py#L20-L22
> 
> Could you try to set the same logic in your usecase. I am going to fix #575 on a follow up PR.您可以尝试在您的用例中设置相同的逻辑吗？我将在后续 PR 中修复 #575 。

Okay, thank you

Hennzau, 2025-03-18:
I'm closing this issue. To recap, the solution is to add a dynamic node that ask for input

closed 2025-03-18 by Hennzau