dora-rs/dora #575: MacOS bug on 0.3.5rc0

Hennzau, 2024-07-02:
**Describe the bug**

When I launch a dataflow where a node is dynamic, I can't stop it; the coordinator/daemon crashes (CTRL+C if attached, dora stop if detached).

**To Reproduce**
Steps to reproduce the behavior:
1. Dora start daemon: `dora up` 
2. Start a new dataflow: `dora start dataflow.yaml` (optional: --detach)
3. Stop dataflow: `dora stop` / `CTRL+C`
At this step it blocks until I do:
6. Destroy dataflow: `dora destroy`

**Environments (please complete the following information):**
 - System info: MacOS
 - Dora version: 0.3.5rc0

Hennzau, 2024-07-02:
on windows I have this error in `dora-coordinator` output file:

```
2024-07-01T19:56:46.554799Z ERROR dora_coordinator::control: failed to send reply
2024-07-01T19:56:46.554844Z ERROR dora_coordinator::control: failed to send reply
```

haixuanTao, 2024-07-03:
I'm unable to reproduce. I get the following behaviour:

https://github.com/dora-rs/dora/assets/22787340/c2a080dd-0295-4a5c-babb-90e7dfc22c90

Are you saying that the dynamic node is not stopping? Or that the dataflow is not stopping?

Hennzau, 2024-07-03:
Hi, it's the dataflow that is not stopping. Even if I don't run the dynamic node, i'm unable to stop the dataflow. Here is a record: 

https://github.com/dora-rs/dora/assets/72349109/3f1e67f0-77e2-45bd-9222-1c56a88e0a9a

haixuanTao, 2024-07-03:
So I think that you might need to start the dynamic node first before stopping the dataflow. I think not starting the dataflow is going to make it impossible to stop a not started dataflow.

I think we should fix this but this might be your error.

phil-opp, 2024-07-10:
I opened https://github.com/dora-rs/dora/pull/583 with a potential fix. Perhaps you could give it a try if you have time, @Hennzau?

Hennzau, 2024-07-11:
> I opened #583 with a potential fix. Perhaps you could give it a try if you have time, @Hennzau?

Hi! It works now! Thanks

PR #583 (2024-07-11): Don't wait for non-started dynamic nodes on stop
This PR removes all pending dynamic nodes on stop instead of waiting for them. This fixes the issue that dataflows with dynamic nodes cannot be stopped properly if the dynamic node was not started yet.

Fixes #575
closed 2024-07-11 by Hennzau