dora-rs/dora #609: dora_next_event()'s delay  is too long

lassloss, 2024-08-05:
Here are some examples of successes and failures during my testing:
1. C++version, transfer 1.3 MB images, use current compiled target/debug/dora to run the project, and the delay in obtaining images is about 100ms when dora_next_event() is used. If using the directly downloaded compiled dora(v0.3.5) to run XXX.yml, the delay is 4ms (only for versions before July 19th).
2. In versions after July 19th, when running the official example using downloaded compiled dora, the second node cannot receive the data from the previous node due to deserialization issues. This may be caused by the mismatch between the cargo install dora cli compiled dora and the latest C++library functions. Simultaneously using current compiled target/debug/dora to run the program successfully, but the latency is still around 100ms.

haixuanTao, 2024-08-05:
So, you have to use the release version if you want to have the best latency.

Make sure to use `--release` when compiling dora.

If you're using the latest main branch to compile C++, you should also use dora main branch to start your dataflow.

Otherwise, use https://github.com/dora-rs/dora/tree/v0.3.5 which is tagged to 0.3.5

lassloss, 2024-08-06:
> If you're using the latest main branch to compile C++, you should also use dora main branch to start your dataflow.

When I use the latest main branch to compile C++, the latency is 100ms. How can I reduce the latency? Should I use --release? I'm not familar with cargo and haven't found where the dora is compiled.  Thanks!

Hennzau, 2024-08-06:
Hi! What you should do is:

- install Dora-cli latest main:
 - clone the Dora directory and launch `cargo install .` inside of it
- build the cxx node api in release mode:
 - inside the repository go to apis/cxx/node, and launch `cargo build --package dora-node-api-cxx --release`

closed 2025-03-18 by Hennzau