dora-rs/dora #610: unaligned shmem access on aarch64

taoliv, 2024-08-06:
**Describe the bug**
I tried to run examples/c-dataflow on aarch64 through shared memory, but c_node, c_link, and runtime_node all crashed. And there is no problem when I run examples/c-dataflow through shared memory on x86_64 or through TCP on aarch64.

The reason is that there are misaligned atomic memory access operations in `shared_memory_Server:: channel:: ShmemChannel:: send_raw`:
```rust
// write len second for synchronization
self.data_len()
    .store(msg.len() as u64, std::sync::atomic::Ordering::Release);
```

**To Reproduce**
Steps to reproduce the behavior:
1. Modify `examples/c-dataflow/dataflow.yml`
```diff
diff --git a/examples/c-dataflow/dataflow.yml b/examples/c-dataflow/dataflow.yml
index 96d5da2f..9c905f6e 100644
--- a/examples/c-dataflow/dataflow.yml
+++ b/examples/c-dataflow/dataflow.yml
@@ -1,3 +1,4 @@
+communication: {_unstable_local: Shmem}
 nodes:
   - id: c_node
     custom:
```
2. Compile dora and examples/c-dataflow to run on aarch64
3. Dora start daemon: `dora up` 
4. Start a new dataflow: `dora start dataflow.yaml`

**Expected behavior**
Dora and nodes exited normally.

**Screenshots or Video**
Dora output:
```
015db1f7-d7c8-7718-9634-c341da4d41c0
attaching to dataflow (use `--detach` to run in background)
ERROR c_node: exited because of signal 7
WARN  c_node: node exited before initializing dora connection
ERROR c_sink: exited because of signal 7
WARN  c_sink: node exited before initializing dora connection
ERROR runtime-node: exited because of signal 7

[ERROR]
Dataflow 015db1f7-d7c8-7718-9634-c341da4d41c0 failed:

Node `c_node` failed: exited because of signal 7
Node `c_sink` failed: exited because of signal 7
Node `runtime-node` failed: exited because of signal 7
```

`out/015db1f7-d7c8-7718-9634-c341da4d41c0/log_c_node.txt`:
```
[c node] Hello World
```

`out/015db1f7-d7c8-7718-9634-c341da4d41c0/log_c_sink.txt`:
```
[c sink] Hello World
```

`out/015db1f7-d7c8-7718-9634-c341da4d41c0/log_runtime-node.txt` is empty.

backtrace:
```
Timestamp:2024-08-01 15:56:23.179
Pid:4669
Uid:0
Process name:/data/local/tmp/dora-debug/dora
Reason:Signal:SIGBUS(BUS_ADRALN)@0x0000007f9e26d111
Fault thread Info:
Tid:4678, Name:dora
#00 pc 00000000018b156c /data/local/tmp/dora-debug/dora(core::sync::atomic::atomic_store::hf218c562153ff115+100)
#01 pc 00000000018b14e4 /data/local/tmp/dora-debug/dora(core::sync::atomic::AtomicU64::store::h20f0671d9c92596d+32)
#02 pc 00000000018b0740 /data/local/tmp/dora-debug/dora(shared_memory_server::channel::ShmemChannel::send_raw::hd0633cba23f25245+312)
#03 pc 0000000000ff91bc /data/local/tmp/dora-debug/dora(shared_memory_server::channel::ShmemChannel::send::hfc482dcea941a91d+264)
#04 pc 0000000000f65d04 /data/local/tmp/dora-debug/dora(shared_memory_server::ShmemClient$LT$T$C$U$GT$::request::h761ba20c3937f1de+40)
#05 pc 0000000000f6d430 /data/local/tmp/dora-debug/dora(dora_node_api::daemon_connection::DaemonChannel::request::h143d45692d041175+100)
#06 pc 0000000000f6cf80 /data/local/tmp/dora-debug/dora(dora_node_api::daemon_connection::DaemonChannel::register::hd8ede7e24195958d+248)
#07 pc 0000000000f5fd3c /data/local/tmp/dora-debug/dora(dora_node_api::event_stream::EventStream::init_on_channel::h321cf26ff6941b9f+312)
#08 pc 0000000000f621fc /data/local/tmp/dora-debug/dora(dora_node_api::event_stream::EventStream::init::h99ad8f1fc3f77a09+3256)
#09 pc 0000000001024f70 /data/local/tmp/dora-debug/dora(dora_node_api::node::DoraNode::init::heaa0b5f8f3c0b02c+2288)
#10 pc 00000000005cf37c /data/local/tmp/dora-debug/dora(dora_runtime::run::_$u7b$$u7b$closure$u7d$$u7d$::_$u7b$$u7b$closure$u7d$$u7d$::hc19627ab9589f4be+1968)
#11 pc 00000000005ce9e8 /data/local/tmp/dora-debug/dora(dora_runtime::run::_$u7b$$u7b$closure$u7d$$u7d$::hea8a72c58dc94895+2636)
#12 pc 0000000000612a80 /data/local/tmp/dora-debug/dora(_$LT$core..pin..Pin$LT$P$GT$$u20$as$u20$core..future..future..Future$GT$::poll::hb5497e2273865863+52)
#13 pc 000000000064bd10 /data/local/tmp/dora-debug/dora(tokio::runtime::scheduler::current_thread::CoreGuard::block_on::_$u7b$$u7b$closure$u7d$$u7d$::_$u7b$$u7b$closure$u7d$$u7d$::_$u7b$$u7b$closure$u7d$$u7d$::h5e64440556fa4fc8+48)
#14 pc 000000000064ba98 /data/local/tmp/dora-debug/dora(tokio::runtime::scheduler::current_thread::CoreGuard::block_on::_$u7b$$u7b$closure$u7d$$u7d$::_$u7b$$u7b$closure$u7d$$u7d$::h5ae30c001faf71cc+264)
#15 pc 0000000000649898 /data/local/tmp/dora-debug/dora(tokio::runtime::scheduler::current_thread::Context::enter::h1b0af5dd2e76bc2a+304)
#16 pc 000000000064ae40 /data/local/tmp/dora-debug/dora(tokio::runtime::scheduler::current_thread::CoreGuard::block_on::_$u7b$$u7b$closure$u7d$$u7d$::h42b01c3e6b23100e+368)
#17 pc 000000000064abd0 /data/local/tmp/dora-debug/dora(tokio::runtime::scheduler::current_thread::CoreGuard::enter::_$u7b$$u7b$closure$u7d$$u7d$::hecfad36a5ae85b2b+40)
#18 pc 0000000000653f64 /data/local/tmp/dora-debug/dora(tokio::runtime::context::scoped::Scoped$LT$T$GT$::set::h0b712d062281f638+160)
#19 pc 00000000006518a8 /data/local/tmp/dora-debug/dora(tokio::runtime::context::set_scheduler::_$u7b$$u7b$closure$u7d$$u7d$::h65f24385aa0e3738+44)
#20 pc 0000000000653384 /data/local/tmp/dora-debug/dora(std::thread::local::LocalKey$LT$T$GT$::try_with::he59cae5a64513b44+336)
#21 pc 0000000000652164 /data/local/tmp/dora-debug/dora(std::thread::local::LocalKey$LT$T$GT$::with::h505b02b170aa929a+28)
#22 pc 0000000000651828 /data/local/tmp/dora-debug/dora(tokio::runtime::context::set_scheduler::hc04028d9582d5b43+52)
#23 pc 000000000064a9c8 /data/local/tmp/dora-debug/dora(tokio::runtime::scheduler::current_thread::CoreGuard::enter::hddfd4afce8cfb16e+344)
#24 pc 000000000064abfc /data/local/tmp/dora-debug/dora(tokio::runtime::scheduler::current_thread::CoreGuard::block_on::h4fcf2b462cefcb9a+28)
#25 pc 0000000000648da4 /data/local/tmp/dora-debug/dora(tokio::runtime::scheduler::current_thread::CurrentThread::block_on::_$u7b$$u7b$closure$u7d$$u7d$::h360597b158581c67+164)
#26 pc 0000000000650b08 /data/local/tmp/dora-debug/dora(tokio::runtime::context::runtime::enter_runtime::h6c0912d2ff5a698d+240)
#27 pc 0000000000648c00 /data/local/tmp/dora-debug/dora(tokio::runtime::scheduler::current_thread::CurrentThread::block_on::h03e4b388b294111e+160)
#28 pc 000000000061c32c /data/local/tmp/dora-debug/dora(tokio::runtime::runtime::Runtime::block_on::h16cde2774c4215e7+196)
#29 pc 00000000005cde20 /data/local/tmp/dora-debug/dora(dora_runtime::main::_$u7b$$u7b$closure$u7d$$u7d$::hea90e65a47b3234b+204)
#30 pc 00000000006055c8 /data/local/tmp/dora-debug/dora(std::sys_common::backtrace::__rust_begin_short_backtrace::h6850b70ace9aaa6b+8)
#31 pc 00000000005dd710 /data/local/tmp/dora-debug/dora(std::thread::Builder::spawn_unchecked_::_$u7b$$u7b$closure$u7d$$u7d$::_$u7b$$u7b$closure$u7d$$u7d$::hfce10b6da7abe9a1+36)
#32 pc 00000000005ffa38 /data/local/tmp/dora-debug/dora(_$LT$core..panic..unwind_safe..AssertUnwindSafe$LT$F$GT$$u20$as$u20$core..ops..function..FnOnce$LT$$LP$$RP$$GT$$GT$::call_once::hc67fc9071f62e266+36)
#33 pc 00000000005fc660 /data/local/tmp/dora-debug/dora(std::panicking::try::do_call::heb1b570fde0eab46+72)
#34 pc 00000000005ff000 /data/local/tmp/dora-debug/dora(__rust_try+24)
#35 pc 00000000005f8800 /data/local/tmp/dora-debug/dora(std::panicking::try::h52f0734d8b93dd34+88)
#36 pc 00000000005dd5b4 /data/local/tmp/dora-debug/dora(std::thread::Builder::spawn_unchecked_::_$u7b$$u7b$closure$u7d$$u7d$::hb391c98882113d92+420)
#37 pc 00000000005dfa54 /data/local/tmp/dora-debug/dora(core::ops::function::FnOnce::call_once$u7b$$u7b$vtable.shim$u7d$$u7d$::h47e772f995e92bf3+16)
#38 pc 0000000001d13ab0 /data/local/tmp/dora-debug/dora(std::sys::unix::thread::Thread::new::thread_start::h4489772baef91967+40)
#39 pc 00000000000fda90 /system/lib/ld-musl-aarch64.so.1
#40 pc 0000000000078f68 /system/lib/ld-musl-aarch64.so.1
```

**Environments (please complete the following information):**
 - System info: [use `uname --all` on Linux]
  ```
  Linux localhost 5.10.165 #1 SMP Fri Jul 26 14:37:48 CST 2024 aarch64
  ```
 - Dora version: [use `dora --version` and `pip show dora-rs`]
  ```
  dora-cli 0.3.5
  ```

**Additional context**
Add any other context about the problem here.

haixuanTao, 2024-08-06:
Ok, interesting!

Do you know how we could fix this?

taoliv, 2024-08-07:
It could be fix if the `Atomic`s are 8-bytes aligned.

phil-opp, 2024-08-07:
Thanks a lot for reporting this! I opened a PR at https://github.com/dora-rs/dora/pull/612 that hopefully fixes this issue. If you have time, could you try whether the PR fixes the issue for you? Thanks!

taoliv, 2024-08-08:
Thanks. I tried this PR, but the address of data_len is still misaligned. I fixed it and left a commit in the PR.

closed 2024-08-09 by haixuanTao