dora-rs/dora #615: Request New Feature: Dora-Flow Concatenation

XiaoKuge, 2024-08-08:
**Is your feature request related to a problem? Please describe.**
Our application (MAE Agent) is essentially a dora flow. We hope to compose multiple applications into a bigger application. So, under the hood, we will need dora-rs to be able to concatenate multiple dora-flow into one. Essentially, we need to concatenate flow1 with flow2, and generate flow3 as output.     

**Describe the solution you'd like**
Provide the function such as: FlowLink(input_flow1.yml, input_flow2.yml, output_flow.yml)

**Describe alternatives you've considered**
N/A

**Additional context**
Note: There maybe m outputs in input_flow1 and n inputs in input_flow2. 

Thank you!

haixuanTao, 2024-08-08:
Unfortunately, we won't provide a way to have concatenation of multiple dataflow in the short term.

We will focus on stabilising the existing API.

XiaoKuge, 2024-08-09:
For your information, we(MAE) have implemented a temporary solution that can concatenate two flows, assuming flow1's last node's output will become flow2's first node's input node.

haixuanTao, 2024-08-09:
ok great! To hear!

closed 2024-08-09 by haixuanTao